KLayout/klayout #955: Build breaks: unable to find library -lklayout_lib

yurivict, 2021-12-26:
```
cd common/ && ( test -e Makefile || /usr/local/bin/qmake-qt5 -o Makefile /disk-samsung/freebsd-ports/cad/klayout/work/klayout-0.27.3/src/plugins/streamers/common/common.pro 'RUBYLIBFILE="/usr/local/lib/libruby27.so.27"' 'RUBYVERSIONCODE="20705"' 'HAVE_RUBY="1"' 'PYTHON="/usr/local/bin/python3.8"' 'PYTHONLIBFILE="/usr/local/lib/libpython3.8.so"' 'PYTHONINCLUDE="/usr/local/include/python3.8"' 'PYTHONEXTSUFFIX=".cpython-38.so"' 'HAVE_PYTHON="1"' 'HAVE_QTBINDINGS="1"' 'HAVE_QT_UITOOLS="1"' 'HAVE_QT_NETWORK="1"' 'HAVE_QT_SQL="1"' 'HAVE_QT_SVG="1"' 'HAVE_QT_PRINTSUPPORT="1"' 'HAVE_QT_MULTIMEDIA="1"' 'HAVE_QT_DESIGNER="1"' 'HAVE_QT_XML="1"' 'HAVE_64BIT_COORD="0"' 'HAVE_QT="1"' 'HAVE_QT5="1"' 'HAVE_CURL="0"' 'HAVE_EXPAT="0"' 'PREFIX="/usr/local"' 'RPATH="/usr/local/lib/klayout"' 'KLAYOUT_VERSION="0.27.3"' 'KLAYOUT_VERSION_DATE="2021-12-25"' 'KLAYOUT_VERSION_REV="d2ad974ff4eb"' 'RUBYINCLUDE="/usr/local/include/ruby-2.7/"' 'RUBYINCLUDE2="/usr/local/include/ruby-2.7//amd64-freebsd13"' 'QMAKE_CXXFLAGS=-O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  -fPIC -I/usr/local/include/qt5  -I/usr/local/include/qt5/QtCore -I/usr/local/include/qt5/QtDesigner -I/usr/local/include/qt5/QtGui -I/usr/local/include/qt5/QtMultimedia -I/usr/local/include/qt5/QtMultimediaWidgets -I/usr/local/include/qt5/QtNetwork -I/usr/local/include/qt5/QtPrintSupport -I/usr/local/include/qt5/QtSql -I/usr/local/include/qt5/QtSvg -I/usr/local/include/qt5/QtTest -I/usr/local/include/qt5/QtWidgets -I/usr/local/include/qt5/QtXml -I/usr/local/include/qt5/QtXmlPatterns ' 'QMAKE_CFLAGS=-O2 -pipe -fno-omit-frame-pointer  -fstack-protector-strong -fno-strict-aliasing ' 'QMAKE_LFLAGS=  -lQt5Core -lQt5Designer -lQt5Gui -lQt5Multimedia -lQt5MultimediaWidgets -lQt5Network -lQt5PrintSupport -lQt5Sql -lQt5Svg -lQt5Test -lQt5Widgets -lQt5Xml -lQt5XmlPatterns  -L/usr/local/lib/qt5 -lpthread -fstack-protector-strong ' ) && gmake -f Makefile all
ld: error: unable to find library -lklayout_lib
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

make environment:
```
INSTALLROOT=/disk-samsung/freebsd-ports/cad/klayout/work/stage  PYTHON_SITELIBDIR=/usr/local/lib/python3.8/site-packages QT_SELECT=qt5 QMAKEMODULES="/disk-samsung/freebsd-ports/cad/klayout/work/klayout-0.27.3/mkspecs/modules:/usr/local/lib/qt5/mkspecs/modules" XDG_DATA_HOME=/disk-samsung/freebsd-ports/cad/klayout/work  XDG_CONFIG_HOME=/disk-samsung/freebsd-ports/cad/klayout/work  XDG_CACHE_HOME=/disk-samsung/freebsd-ports/cad/klayout/work/.cache  HOME=/disk-samsung/freebsd-ports/cad/klayout/work PATH=/disk-samsung/freebsd-ports/cad/klayout/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin NO_PIE=yes MK_DEBUG_FILES=no MK_KERNEL_SYMBOLS=no SHELL=/bin/sh NO_LINT=YES PREFIX=/usr/local  LOCALBASE=/usr/local  CC="cc" CFLAGS="-O2 -pipe -fno-omit-frame-pointer  -fstack-protector-strong -fno-strict-aliasing "  CPP="cpp" CPPFLAGS="-fno-omit-frame-pointer"  LDFLAGS="  -lQt5Core -lQt5Designer -lQt5Gui -lQt5Multimedia -lQt5MultimediaWidgets -lQt5Network -lQt5PrintSupport -lQt5Sql -lQt5Svg -lQt5Test -lQt5Widgets -lQt5Xml -lQt5XmlPatterns  -L/usr/local/lib/qt5 -lpthread -L/disk-samsung/freebsd-ports/cad/klayout/work/klayout-0.27.3/build-release -fstack-protector-strong " LIBS=""  CXX="c++" CXXFLAGS="-O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  -fPIC -I/usr/local/include/qt5  -I/usr/local/include/qt5/QtCore -I/usr/local/include/qt5/QtDesigner -I/usr/local/include/qt5/QtGui -I/usr/local/include/qt5/QtMultimedia -I/usr/local/include/qt5/QtMultimediaWidgets -I/usr/local/include/qt5/QtNetwork -I/usr/local/include/qt5/QtPrintSupport -I/usr/local/include/qt5/QtSql -I/usr/local/include/qt5/QtSvg -I/usr/local/include/qt5/QtTest -I/usr/local/include/qt5/QtWidgets -I/usr/local/include/qt5/QtXml -I/usr/local/include/qt5/QtXmlPatterns "  MANPREFIX="/usr/local" BSD_INSTALL_PROGRAM="install  -s -m 555"  BSD_INSTALL_LIB="install  -s -m 0644"  BSD_INSTALL_SCRIPT="install  -m 555"  BSD_INSTALL_DATA="install  -m 0644"  BSD_INSTALL_MAN="install  -m 444"
```

Version: 0.27.3
OS: FreeBSD 13
The FreeBSD port build breaks this way.

klayoutmatthias, 2021-12-26:
Please provide more information, specifically the linker command that is executed.

And please note that I don't maintain FreeBSD ports myself. I have no way of reproducing the problem.

yurivict, 2021-12-26:
The link line for ```libklayout_lib.so``` is:

```
clang++ -lQt5Core -lQt5Designer -lQt5Gui -lQt5Multimedia -lQt5MultimediaWidgets -lQt5Network -lQt5PrintSupport -lQt5Sql -lQt5Svg -lQt5Test -lQt5Widgets -lQt5Xml -lQt5XmlPatterns -L/usr/local/lib/qt5 -lpthread -fstack-protector-strong -pthread -Wl,-rpath,/usr/local/lib/klayout -Wl,-rpath,/usr/local/lib/qt5 -Wl,-rpath-link,/usr/local/lib/qt5 -shared -Wl,-soname,libklayout_lib.so.0 -o libklayout_lib.so.0.27.3 libForceLink.o libBasic.o libBasicArc.o libBasicCircle.o libBasicDonut.o libBasicEllipse.o libBasicPie.o libBasicRoundPath.o libBasicRoundPolygon.o libBasicStrokedPolygon.o libBasicText.o qrc_libResources.o  -L/usr/local/lib -lz /wrkdirs/usr/ports/cad/klayout/work/klayout-0.27.3/build-release/lib/lib/../../libklayout_gsi.so /wrkdirs/usr/ports/cad/klayout/work/klayout-0.27.3/build-release/lib/lib/../../libklayout_tl.so /wrkdirs/usr/ports/cad/klayout/work/klayout-0.27.3/build-release/lib/lib/../../libklayout_db.so /usr/local/lib/qt5/libQt5PrintSupport.so /usr/local/lib/qt5/libQt5Designer.so /usr/local/lib/qt5/libQt5MultimediaWidgets.so /usr/local/lib/qt5/libQt5Svg.so /usr/local/lib/qt5/libQt5UiTools.a -L/usr/local/lib/qt5 /usr/local/lib/qt5/libQt5Widgets.so /usr/local/lib/qt5/libQt5Multimedia.so /usr/local/lib/qt5/libQt5Gui.so /usr/local/lib/qt5/libQt5Xml.so /usr/local/lib/qt5/libQt5XmlPatterns.so /usr/local/lib/qt5/libQt5Network.so /usr/local/lib/qt5/libQt5Sql.so /usr/local/lib/qt5/libQt5Core.so -lGL                                                                   
```

Subsequent link lines are similar.

```/wrkdirs/usr/ports/cad/klayout/work/klayout-0.27.3/build-release``` is set to be the output directory and all link lines output results there.
All link lines are missing argument -L with this path.

klayoutmatthias, 2021-12-26:
This is not the line where the error happens. This is the line where the missing library is built. I need to know the link command when the error happens.

yurivict, 2021-12-26:
Here is the [full failure log](https://people.freebsd.org/~yuri/klayout-0.27.3_1.log).

klayoutmatthias, 2022-12-06:
Sorry, I lost this thread.
Does the problem still persist? I'd close the ticket otherwise.

Matthias