KLayout/klayout #965: Confusing LEF/DEF import behaviour with symlinks

antonblanchard, 2022-01-06:
If I import a DEF file that is a symlink, the import window shows the LEF files in the same directory as the symlink (making out as if it is going to use them), but it doesn't, doesn't warn that it did not open them, and ultimately fails.

I'm guessing klayout follows the symlink and then looks for LEF files there (based on strace output).

Another point of confusion, it appears that the list of LEF files is not used and it just reads every LEF in the same directory as the DEF 
regardless of the contents.

Testcase: [lefdef-import.tar.gz](https://github.com/KLayout/klayout/files/7819041/lefdef-import.tar.gz)

klayoutmatthias, 2022-01-15:
Thanks for mentioning the symlink topic.

Are you using the "import LEF/DEF" feature? This is kind of redundant because as of now you can simply read DEF like the other layout formats. This function will be removed soon.

The primary way to configure the DEF reader is through "File/Reader Options". When you're using the "Import" function, the import parameters can be configured with the "Options" button in the import dialog.

Here is actually the place where to add LEF files. And here is also the explanation for the "Reads all LEF files" problem:

[image]

In this case the "Read all LEF files in the same directory" is checked. Uncheck this option if you don't want to have this.

Basically, the LEF entries are intended for some technology setup specific LEF files (tech LEF for example). If you're working with technologies, these LEF files are resolved relative to the technology directory and you can store tech specific or PDK LEF files there.

Matthias

klayoutmatthias, 2022-02-11:
Is this issue solved? I'd like to close it.

closed 2022-02-16 by klayoutmatthias

antonblanchard, 2022-02-16:
Thanks @klayoutmatthias I was indeed using the old LEF/DEF import.