KLayout/klayout #969: Pcell do not work for 0.27.6 on REDHAT7 CentOS7

YiLvErZhaoSanDianwei, 2022-01-12:
Dear @klayoutmatthias,

My Pcell code can work for version 0.27.5 on REDHAT7 CentOS7 linux system, however for version 0.27.6, I got following error:
maximumrecursion depth exceeded while calling a Python object in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in
PCellDeclaration_Native.wants_lazy_evaluation in
PCellDeclaretion.wants_lazy_evaluation in

lukasc-ubc, 2022-01-12:
I saw the same error in the Mac OSX build.

RomainAlbert, 2022-01-12:
And I get the same error under windows.

klayoutmatthias, 2022-01-12:
I'm sorry, something is broken here on 0.27.6.

Please use 0.27.5 until I have fixed this issue.

Apologies for this inconvenience.

Best regards,

Matthias

Kazzz-S, 2022-01-12:
Dear @klayoutmatthias,

My PCells are all in **Ruby** and 0.26.7 (Windows, Linux, and Mac) crashed when it just started, opened a new design, or an existing PCell design.

I have (1) reimplemented `wants_lazy_evaluation()` method. Then, when I have (2) incidentally remade the `klayoutrc` file,
things went well on the three different platforms.
Now I get the following warnings on Mac:
```
Warning: QLayout: Attempting to add QLayout "" to QFrame "", which already has a layout
Warning: Populating font family aliases took 210 ms. Replace uses of missing font family "Sans Serif" with one that exists to avoid this cost. 
Warning: QLayout: Attempting to add QLayout "" to QFrame "", which already has a layout
```
Just for your information.

I'll rebuild DMGs if you release the next maintenance release 0.27.7 :-)

Thank you and best regards,
Kazzz-S

klayoutmatthias, 2022-01-12:
I fixed the two QLayout Qt warnigns as well - thanks for reporting.

But you mean, you cannot load a GDS file with 0.26.7 which you have written with 0.27.6? Or is that a typo and you wanted to write "0.27.6"?

Thanks and best regards,

Matthias

Kazzz-S, 2022-01-12:
> I fixed the two QLayout Qt warnigns as well - thanks for reporting.
> 
> But you mean, you cannot load a GDS file with 0.26.7 which you have written with 0.27.6? Or is that a typo and you wanted to write "0.27.6"?
> 
Sorry, just a silly typo ;-(

Kazzz-S

klayoutmatthias, 2022-01-12:
Very good ... I understand better now :)

I'll publish the new release tomorrow with this fix.

Again, sorry for the inconvenience.

Best regards,

Matthias

klayoutmatthias, 2022-01-15:
I'll close this issue as 0.27.7 should fix this problem now.

Thanks for reporting!

Matthias

closed 2022-01-15 by klayoutmatthias