KLayout/klayout #973: Salt manager, error downloading

lukasc-ubc, 2022-01-14:
Dear @klayoutmatthias, 

Several percent of my users have problems downloading using the package manager.  We've seen this for the SiEPIC-Tools and the SiEPIC-EBeam-PDK packages.  The errors occur for larger files - in this case, they are 9 MB in size.  The solutions I've had are:

1. ask people to install the package using a different means (GitHub Desktop with a symbolic link so they can still receive updates).
2. remove the offending large files (which in this case would break the PDK).

This problem seems intermittent.  I wonder if you have any advice?

thank you
Lukas

[image]

klayoutmatthias, 2022-01-15:
Hi Lukas,

actually this seems to be generated by a timeout (the message could be more intuitive). I hardcoded 10 seconds which appeared enough to me. Do you think it's likely that 10 seconds are exceeded?

I can easily extend it to something like 60 seconds. I just wonder if there is an upper limit. I think it's also possible - although harder - to implement infinite timeout with a way to cancel the download.

For my records: code is tlHttpStreamQt.cc, lines 421+

Best regards,

Matthias

lukasc-ubc, 2022-01-17:
Could this be a parameter that can be modified by script config variable, or OS environment, or software configuration window?  

Infinite with cancel would be nicest, but 30-60 s in the meantime would be better perhaps. 

Lukas Chrostowski
Professor, Electrical and Computer Engineering
University of British Columbia
Quantum Matter Institute
Member, College of the Royal Society of Canada
http://siepic.ca Silicon Photonics
http://quantumbc.ca Quantum Computing CREATE program
https://qmi.ubc.ca/about/directory/lukas-chrostowski

klayoutmatthias, 2022-01-17:
Sure, I'll take a look. I have not though 10 seconds would become a problem, but I could confirm that for big files there really is a latency in the order of seconds.

klayoutmatthias, 2022-02-16:
0.27.8 is released which should fix this problem.

closed 2022-02-16 by klayoutmatthias

KristiaNingsih, 2023-05-16:
I have problems downloading using the package manager. We've seen this for the SiEPIC-Tools and the SiEPIC-EBeam-PDK packages. The errors occur for larger files - in this case, they are 9 MB in size. The solutions I've had are:

ask people to install the package using a different means (GitHub Desktop with a symbolic link so they can still receive updates).
remove the offending large files (which in this case would break the PDK).

klayoutmatthias, 2023-05-16:
You mean 60 seconds timeout is not enough?

wschulkins, 2023-09-21:
I am also having issues with the package manager for SiEPIC-Tools. It seems to download all but one of the files, then it fails the whole installation :( 

For reference the error says: 
Error 404: Not Found, fetching https://github.com/siepic/SIEPIC-Tools.git/lvsn/bc/2680/tags/v0.4.0/klayout_dot_config/pymacros/Keybindings/Waveguides/Waveguide to Path %28forget waveguide type%29.lym

lukasc-ubc, 2023-09-21:
It could be because of the filename which contained ( and ).  I changed the filename, and released v0.4.1.  Can you please try again?

wschulkins, 2023-09-21:
Works now! Thank you!

CSG81, 2024-01-10:
Hi,
I get a download issue:
[image]

klayoutmatthias, 2024-01-10:
Hi @CSG81,

this is because GitHub closed the SVN bridge and the package URL was still referring to this feature.

I migrated the package to my backup SVN instance and changed the URL, but this is the last time this will be possible. In the future, the package should use the Git interface and a corresponding URL. From your side this will require a recent KLayout version which supports Git-hosted packages. I'd recommend 0.28.15.

Matthias

Dr-Long-Kahl, 2024-01-26:
Hi there

After installing the Si-Epic Beam-PDK tool, I am unable to verify the Circuit Simulation OPICS: each of the circuits gives me either of the two problems below:

[image]

or 

[image]

Please can you help me with this.
Many thanks.

lukasc-ubc, 2024-01-26:
> Hi there
> 
> After installing the Si-Epic Beam-PDK tool, I am unable to verify the Circuit Simulation OPICS: each of the circuits with the 'opt-in' label gives me either of the two problems below:
> 
> Please can you help me with this. Many thanks.

This is not a KLayout issue, but rather should be reported here: https://github.com/SiEPIC/SiEPIC-Tools/issues

tektronixJBE, 2024-03-21:
Hi,
Can you please help with this error? it is not a timeout, it happens immediately.
Thx.

[image]

klayoutmatthias, 2024-03-21:
I just checked myself - there is no obvious problem with this package.
Are you able to install other Git-based packages - for example "metainfo_ports 0.0.6"?
You can also check the log view (File/Log Viewer) for more details.

Matthias

tektronixJBE, 2024-03-21:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Matthias,

Thx for your email.

No I am not able to install it either. Can it be related to my firewall?

Regards.

***@***.***

Please be advised that this email may contain confidential information. If you are not the intended recipient, please notify us by email by replying to the sender and delete this message.

tektronixJBE, 2024-03-21:
[image]

klayoutmatthias, 2024-03-21:
Firewall may be an issue, yes.
Salt packages are basically just folders stored in KLayout's configuration space. On Windows that is `C:\user\<yourname>\KLayout\salt\<packagename>`. If you are successful installing the package on a different machine outside the firewall, you can basically transfer the folder to the target machine and install the package that way.

Matthias

tektronixJBE, 2024-03-21:
ok thx. i will try that.

sebastian-goeldi, 2024-03-21:
Maybe it's also a permission issue on the folder? Could you check whether the folder Matthias mentioned is write protected by chance?

As it fails immediately that would be my guess, or otherwise as Matthias mentioned a firewall / DNS issue (though the latter is unlikely due to you being able to resolve the saltmine successful).

Also irrelevant but related, completely forgot to make the metainfo_ports 0.28.17+ as I just did with klive (people don't like to update klayout it seems)

tektronixJBE, 2024-03-21:
thx guys. it seems to work.

i downloaded the .zip files from SiEPIC github.com. unzipped them and move the folders to Klayout salt folder; as instructed by Matthias. i see the SiEPIC 0.5.5 Py3 menu and layers (see pics).

[image]
[image]

klayoutmatthias, 2024-03-22:
Very good. 
So write permission does not seem to be the issue, but @sebastian-goeldi thanks for mentioning this. 
If you figure out the reason and you think it is a bug that can be fixed, let me know.

Kind regards,

Matthias

JoshiRaghavendra, 2024-05-12:
Hi Matthias,

I've been trying to add the 'SiEPIC' on the menu tab, as per the given instructions on (https://github.com/siepic/SiEPIC_EBeam_PDK/wiki/Installation-instructions) this page. I do not get any 'SiEPIC' option on the menu tab, but rather a whole lot of other options (as attached in the screenshots). I even tried the alternative method but got the same result. I'd be truly grateful for any help or suggestion. Eagerly waiting for any replies. 
[image]
[image]

Regards,
Raghavendra

lukasc-ubc, 2024-05-12:
@JoshiRaghavendra please move your problem to the SiEPIC-EBeam-PDK github instead: https://github.com/SiEPIC/SiEPIC_EBeam_PDK

JoshiRaghavendra, 2024-05-13:
Sure sir

kasdepopee, 2024-05-24:
Hello, I am getting the following error when trying to install the x-section package. Could you please advise how to install this package?
[image]

klayoutmatthias, 2024-05-24:
I assume you are using an old version of KLayout. You need to use a recent version ( I recommend 0.29.1) as GiHub has closed their SVN bridge. 

But this ticket is closed. Don't expect a response on closed tickets.

kasdepopee, 2024-05-24:
Got it. Thank you very much for answering though. I didn't check if the thread was closed or not. My bad. Will be more careful next time.

klayoutmatthias, 2024-05-24:
No worries, I still found it :) Is the problem fixed?

Matthias

Shehab2121, 2024-09-12:
Hello, I am trying to install the siepec EBeam PDK package and I get this error 

[image]

klayoutmatthias, 2024-09-12:
Thanks for this note. For some reason I still have to debug, this package got lost on the package index among with two other packages. I have restored these entries. Installation should work again.

Thanks,

Matthias

wriddz, 2024-09-18:
Hi,
I am receiving this while installing siepic_tools 0.5.11. I followed the ways discussed above but am still receiving the same error. Could anyone help me out in this please?
[image]

sebastian-goeldi, 2024-09-18:
@wriddz please provide a little bit more info. What is your klayout version? What is the actual error you are getting from downloading (your screenshot exactly cuts this off)?

klayoutmatthias, 2024-09-18:
I confirmed it is basically working currently.

[image]

Problems may be elsewhere. Like your organization does not allow git access from the Windows clients. Getting diagnostics from such network problems is notoriously difficult and I assume that is intentional.

Fallback is always to download the original files as .zip archive and unpacking into the "salt" folder from your KLayout home directory (`c:\users\you\KLayout\salt` on windows). But you will not be able to do updates from there and dependencies need to be installed manually.

Matthias

wriddz, 2024-09-19:
Hi @sebastian-goeldi , I am using the 0.29.6 version of the KLayout. I am hereby attaching the full picture of the error.

Hi @klayoutmatthias , actually I tried the way you mentioned, i.e. downloading the .zip archive and unpacking and moving the "Python_packages_for_KLayout" to the salt folder, but still I was facing the same error. Am I missing something out?

[image]

klayoutmatthias, 2024-09-19:
You do not need to download through the package manager if you have installed it yourself. The package manager basically only downloads the files to c:\home\you\klayout\salt\package-name. If you did the same manually, the package should be there, after you restart KLayout.

wriddz, 2024-09-19:
Okay, thank you.

Shehab2121, 2024-09-19:
Hello @klayoutmatthias, 
Hello Everyone,

I am not sure if here is the right place for trying to solve the error I get when I try to use the packages siepic_tools 0.5.11 and siepic_ebeam_pdk 0.4.11 and lumerical.  

The example that I am trying to do is basically create a new layout with the EBeam technology, go to the "EBeam, Components with models" library, and choose and add a component (for e.g., ebeam _bragg_te1550) in the layout. then in the menu tool SiEPIC 0.5.11 Py3, and under Simulations, Components, I click on Component: show Comapct Model in Lumerical INTERCONNECT. 

It would be great if someone could help me solve this. Thanks.

[image]

sebastian-goeldi, 2024-09-19:
@Shehab2121 this is much more likely either a version problem of siepic wrt to klayout's python, or just a general problem of siepic. The better place to ask is probably either the issues of siepic pdk or tools. But generally speaking unhashable type means that your program is trying to use a mutable object (in this case list) where an immutable one is expected. This can happen if you are using something as a dictionary key or try to cache it, for example. My guess is on the caching in this case, but I am not familiar with siepic.

Lyw12123, 2025-02-14:
> Hi, I am receiving this while installing siepic_tools 0.5.11. I followed the ways discussed above but am still receiving the same error. Could anyone help me out in this please? [image]

carolinelias, 2025-02-20:
Hi,

I have an error while trying to install in Klayout : tools: manage packages : install new package : xsection.
The error is : Error fetching spec file for package 'xsection' : Error 404:Not found, fetching http://sami.klayout.org 

[image]

klayoutmatthias, 2025-02-20:
Hi @carolinelias,

you will need at least version 0.28.13 to access Git-based repositories.

Matthias

carolinelias, 2025-02-21:
Thank you very much for your answer.

I still have an issue while installing xsection:

[image]

klayoutmatthias, 2025-02-21:
Hi @carolinelias,

I just tested the package download on a 0.28.13 installation on Ubuntu 22.04. It works for me. 

Basically I'd recommend 0.29.11 as that is the latest version, but that should not make a difference.

Maybe there was a server glitch. 

If that still does not work, then maybe there is general restriction in your organization to access the Codeberg URL. You can try installing a different package, for example "klive" which is also Git-based, but comes from GitHub. If that works, access may be restricted for you.

Matthias

davidchaozhang, 2025-07-21:
Hi, I am attending the edX course taught by professor Lukas Chrostowski. To design the photonic circuit, I need to download XSection from https://codeberg.org/klayoutmatthias/xsection. I followed the instruction to download from Klayout, it failed. I then go to github download. It failed with the following message: SL certificate problem: unable to get local issuer certificate. Can you help me out? Thanks, David

klayoutmatthias, 2025-07-21:
What is your OS? Can you access the repository using the browser or a local git command line?

davidchaozhang, 2025-07-22:
I am using Windows 11 Enterprise version. I used TortoiseGit. I can try to
use git from a command line.
Thanks,
David

taherehjafari25, 2026-01-28:
Hi , when I try to use examples layout, I got this error. I am not sure how to fix this issue

[image]