KLayout/klayout #975: Stipple Lost on 2nd Display

demisjohn, 2022-01-14:
Some stipples appear to _disappear_ on my 2nd monitor, showing only the outline (frame/line) but no stipple.
Dragging window back to primary built-in monitor causes Stipple pattern to reappear.

Versions: 0.27.4 (latest available on native MacOS Big Sur)

Video:

https://user-images.githubusercontent.com/5370181/149561025-404ca9ee-51ed-46bd-870c-8bd8bacd01c3.mov

klayoutmatthias, 2022-01-15:
Thanks for sharing this video.

I think that's a problem of MacOS and maybe with one screen having a (much?) higher resolution than the other. And the screens and drawing is managed by Qt on which I have little influence on.

On Windows or Linux I have not seen such an issue myself yet.

Matthias

demisjohn, 2022-01-17:
Can you suggest a project/page on which I should post this issue?  Perhaps QT?

klayoutmatthias, 2022-01-17:
I don't think the Qt people would look at KLayout code fix the problem.

I'm afraid there is little chance this gets fixed except by chance. Maybe someone with the proper skills and access to the respective hardware can debug and fix the problem. But I don't have the resources to do so myself.

Matthias

Kazzz-S, 2022-01-17:
Thanks for the video :-)

Even though I do not usually use a second display, I've just tested and reproduced the issue between the two types of displays below.
|Display|Model|Size|Resolution|KLayout & OS|
|:------:|:--------:|:--------------:|:---:|:---:|
A|MacBook Pro Retina|13.3"|2650x1600|0.27.7 on Catalina
B|LG E2350|23"|1920x1080| ditto 

More precisely, 
* Create a window in **A** --> move to **B** --> back to **A** : the same as the video
* Create a window in **B** --> move to **A** --> back to **B**: the issue not observed

Kazzz-S

klayoutmatthias, 2022-01-18:
From what you describe, I assume that some Qt event needs to be processed which tells the application that the main window has been shifted to another display and the image needs to be re-rendered. However, that's a wild guess. I will check if I find any such event on Linux.

demisjohn, 2022-01-18:
IN case it helps, to Matthias' point about resolution, here is the info for my displays:
Built-In Laptop Color LCD:
  Display Type:	Built-In Retina LCD
  Resolution:	2560 x 1600 Retina
  Framebuffer Depth:	24-Bit Color (ARGB8888)

PHL 246V5:
  Resolution:	1920 x 1080 (1080p FHD - Full High Definition)
  UI Looks like:	1920 x 1080 @ 60.00Hz
  Framebuffer Depth:	30-Bit Color (ARGB2101010)
  Adapter Type:	Apple USB-C Digital AV Adapter

  Chipset Model:	Intel Iris Plus Graphics 650
  Type:	GPU
  Bus:	Built-In
  VRAM (Dynamic, Max):	1536 MB
  Vendor:	Intel
  Device ID:	0x5927
  Revision ID:	0x0006
  Metal Family:	Supported, Metal GPUFamily macOS 2

Thanks all for looking into this.

stefanottili, 2022-01-18:
MacBook Air M1 2560x1600 connected to 4k Monitor, display setting to scaled, using // and \\ stipple for the two boxes displays both when created on either display and moved to the other.

demisjohn, 2022-01-19:
When \\\ stipple is missing on 2nd display (Philips 1920 x 1080), choosing `Display > Redraw` does not fix it.  (/// stipple is drawn correctly)

stefanottili, 2022-01-29:
Just to get another datapoint, what happens when you add more layer with different stipple ?

klayoutmatthias, 2022-12-06:
0.28 will have better support for high-DPI displays. Maybe that fixes the problem.

demisjohn, 2023-05-18:
FYI, this problem recurs even on a brand new display, 
on a new Acer XZ320Q P, via Belkin AVC008 USB-C--> HMDI dock:
  Resolution:	1920 x 1080 (1080p FHD - Full High Definition)
  UI Looks like:	1920 x 1080 @ 60.00Hz
  Framebuffer Depth:	30-Bit Color (ARGB2101010)
  Display Serial Number:	130107C403W01
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Adapter Type:	DVI or HDMI

KLayout version 0.27.11

stefanottili, 2023-06-16:
Since Matthias stated “0.28 will have better support 0.28 will have better support for high-DPI displays. Maybe that fixes the problem.”, could you give the latest and greatest a try and report back ?

demisjohn, 2024-10-22:
This is fixed, tried on v0.29.7 - stipple is now present on patterns where it used to disappear on 0.27.x!  Nicely done, and thank you!
OK to close this issue.

closed 2024-10-22 by demisjohn