KLayout/klayout #978: question about pt1-pt2

YiLvErZhaoSanDianwei, 2022-01-17:
Hi @klayoutmatthias ,
I find if both pt1 and pt2 are defined as DPoint, pt1-pt2 will return a vector, not a DPoint. however pt1+-pt2 will return a DPoint.
May I know why it is set in this way?

Thanks.

klayoutmatthias, 2022-01-17:
Is this a bug? If that is a general question, a better place to ask it is the forum (https://www.klayout.de/forum).

A general discussion about points and vectors can be found here: https://www.klayout.de/doc-qt4/programming/geometry_api.html#h2-32

Your specific question arises from the definition of the unary "-" operator. "-p" is not defined as "0-p" as you seem to expect but it is rather defined as "(-1)*p". This renders a point mirrored at the origin. Point plus point again gives a point. Basically this operation is not present in the formal framework of "anchored" points and "floating" vectors (the difference is the behaviour upon a translation transformation). But point addition is handy, for example to compute the midpoint of p1 and p2: "(p1+p2)/2" while the formally correct way was "p1+(p2-p1)/2". So eventually "p1+-p2" is a point plus point addition.

YiLvErZhaoSanDianwei, 2022-01-18:
Hi @klayoutmatthias ,
Actually, I want to find the mirror point of p1 refer to p2, so I simply use p3=p2*2-p1, then find p3 is a vector, not a DPoint, so I am thinking it will be easier if DPoint1-DPoint2 also return a DPoint. currently I use p3 =p2*2+-p1, then p3 is a DPoint.

Thank you very much!

closed 2022-01-18 by klayoutmatthias