KLayout/klayout #984: Crash in debug mode when placing a "slow cell"

iqmtestd, 2022-01-23:
Hi @klayoutmatthias,

Klayout crashes in debug mode when a drag-and-drop operation is to fast compared to the pcell's runtime. 

To reproduce press down the mouse button on a "slow cell", drag it to the  layout and release the button before the cell appears. I've added a sleep call to the `Circle` pcell sample to make it slow enough. This also happens with the latest KLayout compiled from master.

An obvious workaround is to be patient with the mouse button :)

Thanks,
David

[image]

klayoutmatthias, 2022-01-23:
Thank you for reporting this issue. 

I can reproduce it, but a quick debugging did not reveal an obvious reason. The crash happens somewhere inside Qt. I'll try to further debug the problem.

I don't see it happening with the debugger open, so a quick workaround is not to use drag and drop, but use the toolbar's "instance" mode to create an instance with the debugger open.

Matthias

klayoutmatthias, 2022-01-30:
Update: apparently that is a problem in Qt 5.12.8 (or other versions). A pointer is set to null, but it's not tested before accessing it.

The workaround I found is to disable the Ruby or Python debugger during drag operations - this means you will not be able to debug PCell code when you drag & drop a PCell into the window. I think this makes even some sense as during drag and drop the mouse cursor is captured by the system. Accessing the debugger is not really in line with the UI system. I guess this is also one of the reasons why the mentioned pointer becomes null.

klayoutmatthias, 2022-02-16:
0.27.8 is released which should fix this problem.

closed 2022-02-16 by klayoutmatthias