KLayout/klayout #985: DRC problem on m1-m1 spacing

tuohangzeng, 2022-01-28:
Hello,

When I am using Klayout to run DRC on a design, the run outputs the following error:

> `m1.3ab : min. 3um.m1 spacing m1 : 0.28um`

signaling that the following lines are finding DRC violations (line 477 on the attached lydrc file)

> `(huge_m1.separation(non_huge_m1, 0.28, euclidian) + huge_m1.space(0.28, euclidian)).output("m1.3ab", "m1.3ab : min. 3um.m1 spacing m1 : 0.28um")`

This is one net, and the two met1 shapes touch each other in the layout, as shown in the screenshot below.
[image]

How should this be resolved? The lydrc script is attached here [sky130A.lydrc](https://github.com/KLayout/klayout/files/7955346/sky130A.txt). I am running this on KLayout 0.27.7 on a Windows 11 (64-bit) system.

Thanks!

klayoutmatthias, 2022-01-29:
I think that the lower vertical part is classified as non-huge, while the upper part is classified as huge here:

```
huge_m2 = m2.sized(-1.5).sized(1.5)
non_huge_m2 = m2 - huge_m2
```

This is trigger a separation violation because both parts are closer than 0.28 µm in this statement:

```
huge_m2.separation(non_huge_m2, 0.28, euclidian)
```

Basically the solution is to check huge_m2 against all other m2. This would also include the second part of the same check:

```
# replace
#   non_huge_m2.space(0.14, euclidian).output("m2.2", "m2.2 : min. m2 spacing : 0.14um")
# by this (READ BELOW):
huge_m2.separation(m2, 0.28, euclidian)
```

But there is a pitfall: as m2 includes huge_m2, a feature called "shielding" will prevent the separation check from triggering (see here for details: https://www.klayout.de/doc-qt5/manual/drc_runsets.html#k_11)

Fortunately (from 0.27 on) you can turn off this feature with the "transparent" flag, so the solution is:

```
huge_m2.separation(m2, 0.28, transparent, euclidian).output("m2.3ab", "m2.3ab : min. 3um.m2 spacing m2 : 0.28um")
```

Regards,

Matthias

tuohangzeng, 2022-01-29:
Thank you very much!

Best,
Tuohang

closed 2022-01-29 by tuohangzeng