KLayout/klayout #988: ERROR: wrong -pyinc path:

jimbo1990, 2022-02-02:
Hello, I am trying to compile k-layout build however I keep getting the following error:

**ERROR: wrong -pyinc path: /usr/include/python3.6m/Python does not exist or is not a file**

Im kind of new to linux so Im not sure how to fix this issue if anyone has come across it before please let me know. i will leave the logs for the istallation below.

Thanks

`Scanning installation ..

Version Info:
    Version: 0.28
    Date: 2022-02-02
    Revision: LatestSourcePackage

Using qmake: qmake-qt4

Using Ruby interpreter: ruby
    Ruby library found: /usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5.1
    Ruby headers found: /usr/include/ruby-2.5.0 and /usr/include/x86_64-linux-gnu/ruby-2.5.0
    Ruby installation is in:
    - /usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5.1 (lib)
    - /usr/include/ruby-2.5.0 (headers)
    - /usr/include/x86_64-linux-gnu/ruby-2.5.0 (arch headers)
    Ruby version code is 20501

Using Python interpreter: python3
    INFO: Python library not in default path, trying to use MULTIARCH
    Python library found: /usr/lib/x86_64-linux-gnu/libpython3.6m.so
    Python headers found: /usr/include/python3.6m
    Python extension suffix: .cpython-36m-x86_64-linux-gnu.so
    Python installation is in:
    - /usr/lib/x86_64-linux-gnu/libpython3.6m.so (lib)
    - /usr/include/python3.6m (includes)

Features:
    Qt bindings enabled
*** ERROR: wrong -pyinc path: /usr/include/python3.6m/Python does not exist or is not a file
    - `

klayoutmatthias, 2022-02-03:
I think you have not installed the Python3 development files. On CentOS8 they can be found in "python3-devel", on Ubuntu that is "python3-dev". For other system, there may be similar packages.

Matthias

klayoutmatthias, 2022-02-16:
Is that issue solved? I'd like to close this one.

closed 2022-02-16 by klayoutmatthias