KLayout/klayout #990: klayout-0.27.3 fails to build with musl

akhuettel, 2022-02-06:
(Note, this is an issue that has been found via automated build testing. I'm not a musl expert and not running the CI system either...)

In an installation using the alternative libc [musl](https://musl.libc.org/), the build of klayout-0.27.3 fails with the following error (which may only be the first one):

`../../src/lay/lay/laySignalHandler.cc:44:12: fatal error: execinfo.h: No such file or directory`

More details including a full build log can be found [on the original bug](https://bugs.gentoo.org/832743).

klayoutmatthias, 2022-02-06:
Is there is no `execinfo.h` available, you can try to exclude the stack trace code with some proper `#if defined(...)` in `laySignalHandler.cc`.

akhuettel, 2022-02-26:
PS. An alternative would be to use [https://github.com/ianlancetaylor/libbacktrace](libbacktrace) when glibc is not available.
(I just added that to Gentoo, because this problem occurs more often, not just for klayout)

klayoutmatthias, 2022-12-06:
Any news here? Otherwise I'd close this ticket.
Matthias

akhuettel, 2023-06-12:
> Any news here? Otherwise I'd close this ticket. Matthias

Only that in the meantime I figured out that the libbacktrace mentionend above is *not* a valid replacement.
Feel free to close. I still have my Gentoo bug, and may poke at it if I ever need a useful occupation.

(There's to my knowledge no actual user involved here (yet), this came up during automated build testing.)