KLayout/klayout #995: pcell-within-library-appears-as-new-independant-topcell-in-the-cells-side-bar-in-layout

joelcollet, 2022-02-10:
Hello Matthias, all

i dunno how to put an explicit title on the behaviour i seen, nor found an answer to what i see in the forum, so sorry if this behaviour is intended and/or documented.
Actually, i make "huge" use of the library concept and of the Pcells. I recently upgraded from a rather old version (0.25..) and i have seen a behaviour appear in the cells side bar view. Let me explain:
Make a gds library called, say test.gds with a library title test_lib. Put in this library in a topcell (lets call it test) some standard patterns and somme Pcells (basic layout objects will do it, a circle for example) save this gds, close klayout  and reopen it . make a new layout and then put an instance of the test libray in it. put other patterns (or not, doesn't matter)
you will have in the cells side bar view something like

Topcell
> Test_lib.test
     > test_lib.Basic.CIRCLE(xxx)

which is what we can see in the old version.
then save and close this layout
If you simply reopen this layout, instead of having the view as above (which is what i expect and what i got with the "old" version, you will have

Topcell
> Test_lib.test
     > test_lib.Basic.CIRCLE(xxx)
Basic.circle(xxx)

as you can notice the Pcell cell appear in the cell views as a "fake" or "ghost" independant topcell which can be for example , viewed, deleted, copied, etc

As soon as you try to put a new object instance (can be a pcell or a library pattern) these "independant topcell" disappear from the view.

if you delete them then reopen the saved file they do "magically" reappear...

So i dunno if is intended or not , or if there is a way to not have this feature but  it make the cell tree view rather hard to "decode" and review...

any clue ?

Regards
Joël

klayoutmatthias, 2022-02-11:
Hi Joël,

such top cells temporarily appear as "orphan" PCells, for example while you are dragging a PCell to it's destination. They should be cleaned up regularly. This is why it disappears. I don't know why it is created during the load. I need to reproduce that problem.

Matthias

joelcollet, 2022-02-11:
Hi Matthias

Yes, that is what i finally guess‌ed when seeing that the behaviour was similar to when creating a new pcell instance.

if it matters/help, i am on windows 10, with the 64 bit version.

Thanks
Joël
 

De : "Matthias Köfferlein"
A : "KLayout/klayout" ,"joelcollet" ,"Author"
Envoyé: vendredi 11 Février 2022 15:26
Objet : Re: [KLayout/klayout] pcell-within-library-appears-as-new-independant-topcell-in-the-cells-side-bar-in-layout (Issue #995)
 

 

Hi Joël,

such top cells temporarily appear as "orphan" PCells, for example while you are dragging a PCell to it's destination. They should be cleaned up regularly. This is why it disappears. I don't know why it is created during the load. I need to reproduce that problem.

Matthias

closed 2022-02-12 by klayoutmatthias

joelcollet, 2022-05-11:
Hello Matthias

Sorry for the late answer. Was 3 really bad monthes for me since!
This is working fine now.
thanks for the quick reply and support!
Regards

Joël

commit 60aab8b3 (2022-02-12): Fixed #995 (new topcell after load) (#1000)
commit 1bdf045c (2022-02-12): Fixed #995 (new topcell after load) (#1000)