KLayout/klayout #1011: DRC in deep mode highlights erroneous width violation with text

dansharkey, 2022-02-23:
There is an issue with DRC checks where a "text" object will cause a violation of a width check if it is over the top of a cell when DRC is run in "deep"/hierarchical mode. I have written a short macro which generates a layout which suffers from the problem, along with a DRC script which highlights the issue.

Macro to create layout which exhibits the issue:
``` ruby
# save this script to "generate.rb" and run it with "klayout -r generate.rb -b"
layout = RBA::Layout::new

# database unit 1nm:
layout.dbu = 0.001

# create a top cell
top = layout.cell(layout.add_cell("TOP"))
# create a subcell
subcell = layout.cell(layout.add_cell("Subcell"))

# create a layer: layer number 1, datatype 0
layer = layout.insert_layer(RBA::LayerInfo::new(1, 0))

# create one rectangle with lower-left coordinates at 0,0
# and width of 1um and height of 1um (coordinates are in
# database units)
subcell.shapes(layer).insert(RBA::Box::new(0, 0, 1000, 1000))

top.shapes(layer).insert(RBA::Text::new("Test Text",0,0))

top.insert(RBA::CellInstArray::new(subcell.cell_index, RBA::Trans::new(0,0)))

# write to x.oas
layout.write("x.oas")
```

DRC script to reproduce issue:
``` ruby
deep

layer = input(1)
layer.width(0.5.micron).output(100,0)
```

KLayout version tested: 0.27.7
OS: Windows 10 64-bit
2022-01-12 r428d0fe8c

klayoutmatthias, 2022-02-23:
Many thanks for this perfectly prepared test case.

I see what's the problem. Texts are turned into 2x2 nm "dots". This was once a way to implement text/polygon interaction for net name annotation in LVS. I though actually I had reverted this, but I need to check that. In any case, this behaviour does not make a lot of sense.

The problem should not happen if the text is on (rather than on the edge of) the polygon. 

An easy workaround is to use "polygons" instead of "input":

```
deep

layer = polygons(1)
layer.width(0.5.micron).output(100,0)
```

This will declare the input "polygon only" and texts are ignored.

Matthias

dansharkey, 2022-02-24:
Brilliant Matthias, many thanks for the workaround - I can confirm it works well and has fixed my immediate issue.

klayoutmatthias, 2022-04-24:
Solution released with 0.27.9

closed 2022-04-24 by klayoutmatthias