KLayout/klayout #1015: DRC script disappears from script list due to "&" character

chkkbim, 2022-03-01:
[1] Using KLayout 0.27.8 64bit on Windows 11.

**[2] Background**
I edit my DRC script in an external text editor, while keeping the script open in KLayout Macro Editor.
When I save and switch from the text editor to the Macro Editor, the Editor prompts me to click OK to refresh.
So far so good.

**[3] Issue**
When I type a "&" character in the script in the text editor, then save and switch to the Macro Editor, the script disappears from the script list on the left side of the Macro Editor.
This happens even if there is a "#" character at the beginning of the line.

**[4] To reproduce issue**

```
report("test123")
nwell = polygons(255, 0)
# width(nwell, 30.um).output("Nwell and width", "Min width = 3 um")
```

First type the above 3 lines into the Macro Editor, then save.
Then open the script in an external text editor, replace "and" (in 3rd line) by "&", then save.
Switch back to the Macro Editor, click "OK" when asked to refresh.

Kazzz-S, 2022-03-01:
A *.lydrc file is an XML file.

So, an ampersand `&` symbol has to be escaped with `&amp;`.
Otherwise, a syntax error causes the script to disappear from the script list.

Create the three lines below using an external editor,
```
report("test123")
nwell = polygons(255, 0)
# width(nwell, 30.um).output("Nwell & width", "Min width = 3 um")
```
Next, copy and paste them into a new macro editor pane, then save them as a new DRC script.
KLayout will escape the `&` symbol appropriately.

klayoutmatthias, 2022-03-01:
@Kazzz-S is correct. If you edit in an external editor, you need to make sure the file is valid XML.

If you want DRC scripts as plain text files, use extension ".drc".

Matthias

chkkbim, 2022-03-02:
I will use the ".drc" solution.
Thanks !