KLayout/klayout #1021: LVS unexpectedly failing

FatsieFS, 2022-03-09:
Here is a test case of an SRAM array: [sramlvs.tar.gz](https://github.com/KLayout/klayout/files/8214709/sramlvs.tar.gz)
There is a run script to run the lvs on a 2X2 and a 2X4 array; in the 2X4 array the 2X2 array is a subcell.
For the 2X2 array the lvs is OK but for the 2X4 array it fails and I don't see why.
I am using klayout version 0.27.8

FatsieFS, 2022-03-09:
Also tried now without the `flat` line but it gives the same result: 2X2 OK, 2X4 fails.

klayoutmatthias, 2022-03-10:
I had a brief look an reproduce the problem.

The LVS script has a couple of problems that prevent true hierarchical mode. I was able to modify it so I can achieve match in deep mode (I will tell you how), but the flat problem remains (BTW: "flat" is the default, so adding it does not make a difference).

I need to debug the issue.

Matthias

klayoutmatthias, 2022-03-12:
I have fixed the problem: "consider net names" wasn't fully effective in that case and this created a conflict. This is going to be released in version 0.27.9.

If you use "consider_net_names(false)" the LVS matches. However I'd not recommend that as a workaround as then LVS assignes the wrong bitline polarities due to the symmetry of the latter. Using labels is much better in fact.

Another workaround is to use deep mode and hierarchical matching. As mentioned, several issues prevent the current LVS from running in deep mode and some changes are required:

1. The substrate does not need to be a solid plate. This will only force flat evaluation. There is "connect_global" for the purpose of generating "overall" nets. Still, "substrate__Sky130" is required as a receiver for the device terminals. It can be empty however:

```
substrate__Sky130 = polygon_layer
```

2. Using the substrate in booleans is not required and forces flat evaluation, so these booleans can be rewritten:

From:

```
gate__mosgate = (difftap&poly__conn&(extent.sized(0.34)-hvi))
gate__mosfet__nfet_01v8 = (gate__mosgate&substrate__Sky130&nsdm)
gate__mosfet__nfet_01v8_lvt = (gate__mosgate&substrate__Sky130&nsdm&lvtn)
gate__mosfet__nfet_g5v0d10v5 = (gate__hvmosgate&substrate__Sky130&nsdm)
```

To:

```
gate__mosgate = (difftap&poly__conn-hvi)
gate__mosfet__nfet_01v8 = (gate__mosgate&nsdm)
gate__mosfet__nfet_01v8_lvt = (gate__mosgate&nsdm&lvtn)
gate__mosfet__nfet_g5v0d10v5 = (gate__hvmosgate&nsdm)
```

3. The following connect statements are converted into global ones:

```
connect(substrate__Sky130, extent.sized(0.34))
```

into:

```
connect_global(substrate__Sky130, "vss")
```

and

```
connect(difftap__conn__psdm, substrate__Sky130)
```

into:

```
connect_global(difftap__conn__psdm, "vss")
```

4. The SP6TArray_2X1 cell has two vdd pins in layout, but one in schematic, so this implicit connect is required to fix that:

```
connect_implicit("SP6TArray_2X1", "vdd")
```

(without a cell, "connect_implicit" is only effective at top level)

5. The SRAM cell's drains overlap across cell instances, so we need to introduce that cheat to tell KLayout not to propagate the devices up in the hierarchy (see https://www.klayout.de/doc-qt5/about/drc_ref_global.html#k_17 about cheats):

```
cheat("SP6TCell") do
    extract_devices(mos4("sky130_fd_pr__nfet_01v8__model"), {
        "SD" => difftap__conn, "G" => gate__mosfet__nfet_01v8, "tG" => poly__conn, "W" => substrate__Sky130,
[... 206 lines cut ...]
# pfet_01v8_hvt
extract_devices(mos4("sky130_fd_pr__pfet_01v8_hvt__model"), {
    "SD" => difftap__conn, "G" => gate__mosfet__pfet_01v8_hvt, "tG" => poly__conn, "W" => nwm,
})
# pfet_01v8_lvt
extract_devices(mos4("sky130_fd_pr__pfet_01v8_lvt__model"), {
    "SD" => difftap__conn, "G" => gate__mosfet__pfet_01v8_lvt, "tG" => poly__conn, "W" => nwm,
})
# pfet_g5v0d10v5
extract_devices(mos4("sky130_fd_pr__pfet_g5v0d10v5__model"), {
    "SD" => difftap__conn, "G" => gate__mosfet__pfet_g5v0d10v5, "tG" => poly__conn, "W" => nwm,
})

netlist

align
ok = compare
if ok then
    print("LVS OK\n")
else
    abort "LVS Failed!"
end
```

Matthias

klayoutmatthias, 2022-04-24:
Solution released with 0.27.9

closed 2022-04-24 by klayoutmatthias