KLayout/klayout #1022: Transient segfault during DRC

mguthaus, 2022-03-09:
I have a segfault during DRC with the sky130 technology on v0.27.8. I cannot consistently replicate it... However, does this give any hint what it could be:

```
ERROR: Signal number: 11
Address: 0x198
Program Version: KLayout 0.27.8 (2022-03-06 rcbd805f8e)

Backtrace:
/usr/local/klayout/libklayout_lay.so.0 +0x2cfde7 lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x46210 killpg [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x6c8e44 db::Layout::update() const [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x5f0061 db::Cell::begin_parent_cells() const [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x131a73a db::local_processor<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_tr
ans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::compute_results(db::local_processor_contexts<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >
, db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >&, db::local_operation<db::polygon_ref<db::polygon<int>
, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > > const*, std::vector<unsigned in
t, std::allocator<unsigned int> > const&) const [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x131ad72 db::local_processor<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_tr
ans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::run(db::local_operation<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db:
:polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >*, unsigned int, std::vector<unsigned int, std::allocator<unsigned int> > co
nst&, std::vector<unsigned int, std::allocator<unsigned int> > const&) [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x131b013 db::local_processor<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_tr
ans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::run(db::local_operation<db::polygon_ref<db::polygon<int>, db::disp_trans<int> >, db::polygon_ref<db:
:polygon<int>, db::disp_trans<int> >, db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >*, unsigned int, unsigned int, std::vector<unsigned int, std::allocator<unsi
gned int> > const&) [??:?]
/usr/local/klayout/libklayout_db.so.0 +0x133467d db::DeepRegion::selected_interacting_generic(db::Region const&, int, bool, db::InteractingOutputMode, unsigned long, unsig
ned long) const [??:?]
/usr/local/klayout/libklayout_db.so.0 +0xf22d22 db::Region::selected_interacting(db::Region const&, unsigned long, unsigned long) const [??:?]
/usr/local/klayout/libklayout_db.so.0 +0xf406d0 void std::vector<db::Region, std::allocator<db::Region> >::emplace_back<db::Region>(db::Region&&) [??:?]
/usr/local/klayout/libklayout_rba.so.0 +0x278a5 rba::RubyInterpreter::add_package_location(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >
const&) [??:?]
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x20e026 rb_dtrace_setup [??:?]
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21b30e rb_vm_invoke_bmethod [??:?]
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x201406 rb_f_notimplement [??:?]
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x2120ba
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x218870 rb_vm_exec
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21b78e
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x227b33
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x20e026
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21adab
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x201406
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x2120ba
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x218870 rb_vm_exec
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21c918
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21cdc3 rb_vm_call0
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21d19c rb_vm_call_kw
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21f408 rb_funcallv
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0xac5d3 rb_protect
/usr/local/klayout/libklayout_rba.so.0 +0x4577d
/usr/local/klayout/libklayout_rba.so.0 +0x48a18
/usr/local/klayout/libklayout_gsi.so.0 +0x91765
/usr/local/klayout/libklayout_gsi.so.0 +0x91a0e
/usr/local/klayout/libklayout_tl.so.0 +0xb9660 _ZN2tl10Executable10do_executeEv
/usr/local/klayout/libklayout_lym.so.0 +0x48caf _ZN3lym16MacroInterpreter13execute_macroEPKNS_5MacroE
/usr/local/klayout/libklayout_lym.so.0 +0x4fce6 _ZNK3lym5Macro3runEv
/usr/local/klayout/libklayout_lay.so.0 +0x149a70 _ZN3lay15ApplicationBase3runEv
/usr/local/klayout/klayout +0x4a9d
/usr/local/klayout/libklayout_rba.so.0 +0x23bd4
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x20e026
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x21adab
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x201406
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x2120ba
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0x218870 rb_vm_exec
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0xa9491
/lib/x86_64-linux-gnu/libruby-2.7.so.2.7 +0xae2ee ruby_run_node
/usr/local/klayout/libklayout_rba.so.0 +0x24084 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/usr/local/klayout/klayout +0x46fa
/usr/local/klayout/klayout +0x3edd
/lib/x86_64-linux-gnu/libc.so.6 +0x270b3 __libc_start_main
/usr/local/klayout/klayout +0x3f9e
```

mguthaus, 2022-03-10:
I tried rerunning everything with the debug enabled klayout, but it doesn't run into this problem. :(

klayoutmatthias, 2022-03-12:
Hi Matt,

I haven't seen this problem myself yet. The stack trace does not fully make sense, but that doesn't necessarily mean something.

Maybe you can give some details. I assume you run the DRC on the command line (something like "klayout -b -r script.drc"), right? And where can I find the Skywater 130 DRC you're using?

Thanks,

Matthias

klayoutmatthias, 2022-12-06:
You should try 0.28 where a lot of debugging has happened.