KLayout/klayout #1025: automatically start macro when opening klayout

joamatab, 2022-03-12:
with latest klayout versions 

how can we make sure we start a macro automatically when klayout starts?

see code here
https://github.com/gdsfactory/gdsfactory/blob/master/gdsfactory/klayout/pymacros/klive.lym

` <autorun>true</autorun>` used to work in previous klayout versions

https://github.com/gdsfactory/gdsfactory/issues/230

klayoutmatthias, 2022-03-12:
"autorun" still works. There was no change.

If I copy the file into "~/.klayout/pymacros" it is executed on startup.

joamatab, 2022-03-16:
thank you Matthias,

I figured out the issue, the error was coming from klayout not starting correctly due to some macros requiring numpy

I'll open a new ticket for this