KLayout/klayout #1035: 2.5 D view export as image and blender file

tisoso, 2022-03-16:
It would be great if there was an option to export the views from the 2.5D viewer.
1. As image file (png, or jpg)
2. Blender file for further editing in Blender (could also be an stl. or some other 3d file)

klayoutmatthias, 2022-03-16:
The 2.5d view is not called "two and a half d" for no reason. It's not a 3d model. Visualization is different from creating an STL solid for example. The OpenGL rendering pipeline does not care about internal faces and the z buffer will provide face clipping. Generating a 3d solid is something different and more challenging, specifically as most consumer tools of 3d solids are very picky.

joamatab, 2022-03-17:
You can try using trimesh

https://gdsfactory.github.io/gdsfactory/notebooks/03_layers.html#3D-rendering