KLayout/klayout #1039: Issue with code freezes/crashes Macro development GUI when running code to create a polar array

JT-24, 2022-03-17:
Hi all, 

I am new to coding in python and creating PCells in klayout, so apologies in advance for any gaps in my understanding. 

I have tried to create a PCell to generate a polar array of a square or circle feature shape based on user input. Though at present when I try to create the PCell, the typical window which prompts the user to enter the parameters for the PCell does not open. Also, an "Exception Caught" window presents itself (in the Macro Development Window) and I cannot select any of the options presented (see screenshot below): OK, Ignore or Cancel. The GUI freezes and I have to close the program using Task Manager. I am using Windows V10.0.19044. 

[image]

I have included the full code below for reference. I was hoping to debug and fix the code myself but this is near impossible as the GUI keeps crashing before I can navigate to and inspect the source of the issue. If anyone has any ideas on how I can correct this freezing issue and/or the code below, that may help resolve this issue entirely. Thanks!

`import pya

# PCell template
# This macro template provides the framework for a PCell library

# It is recommended to put PCell code into namespaces.
# TODO: change the module name

# The PCell declaration
# Each PCell must provide a declaration. It is recommended to use the PCell name as the class name.
# TODO: change the class name
class PolarArray(pya.PCellDeclarationHelper):

  def __init__(self):

    # Important: initialize the super class
    super(PolarArray, self).__init__()

    # declare the parameters
    self.param("lay", self.TypeLayer, "Layer", default = pya.LayerInfo(5, 0))
    self.param("s", self.TypeShape, "",)
    self.param("n",self.TypeInt,"Number of instances in polar array",default = 4)
    self.param("npoints",self.TypeInt,"Number of points in the circle", default = 64)
    self.param("w",self.TypeDouble,"Width/Diameter",default = 0.5)
    self.param("r",self.TypeDouble,"Radius",default = 0)
    self.param("spacing",self.TypeDouble,"Spacing",default = 0.5)
    self.param("a",self.TypeDouble,"Angle",default = 0)
    self.param("featureshape",self.TypeString,"Feature Shape",default = "Circle")
    self.param("patternwidth",self.TypeDouble,"Pattern Width", default = 100)

  def display_text_impl(self):
    # Provide a descriptive text for the cell
    return "Polar Array(L=" + str(self.lay) , str(self.featureshape), "PW = " + ('%.3f' %self.patternwidth), " FW=" + ('%.3f' % self.w), "S=" + ('%.3f' %self.spacing) + ")"
  
  def coerce_parameters_impl(self):
    # TODO: use x to access parameter x and set_x to modify it's value 
    pass
    if self.lay != 5:
      raise(RuntimeError("layer 5 must be used for membrane patterns!"))
    if self.w <= 0:
      raise(RuntimeError("Feature width must be larger than 0!"))
    if self.r != 0: 
      raise(RuntimeError("Initial radius must be equal to 0!"))
    if self.spacing < self.w:
      raise(RuntimeError("Feature spacing must be larger than feature width!"))
    if self.a != 0:
      raise(RuntimeError("Initial angle must be equal to 0!"))      
    if self.featureshape != "Circle" or self.featureshape != "Square":
      raise(RuntimeError("Feature shape may be a Circle or Square only!"))
    if self.patternwidth < self.w:
      raise(RuntimeError("Pattern width must be larger than feature width!"))
    if self.n > 360:
      raise(RuntimeError("Cannot exceed 360 instances in a polar array!"))
    if self.npoints < 4:
      raise(RuntimeError("At least 4 points are required for a single feature shape"))
     
  def produce_impl(self):
    # TODO: produce the cell content 
    # i.e. self.cell().shapes(self.l_layer).insert(pya.Polygon(...))

     r_dbu = self.r / self.layout.dbu
     w_dbu = self.w/ self.layout.dbu
     spacing_dbu = self.spacing/ self.layout.dbu
     patternwidth_dbu = self.patternwidth/ self.layout.dbu
    
    #Insert the given shape at the origin with the desired feature width
     if self.featureshape == "Circle":
       pts = [] #List to house all the points of the circle co-ordinates
       da = math.pi * 2 / self.npoints #delta in angle between points in the circle feature    
       # compute the circle
       for i in range(0, self.npoints):
          pts.append(pya.Point.from_dpoint(pya.DPoint((w_dbu/2) * math.cos(i * da), (w_dbu/2) * math.sin(i * da)))) #Adding the co-ordinates to the pts list
          # create the shape
          self.cell.shapes(self.lay_layer).insert(pya.Polygon(pts))
      
     elif self.featureshape == "Square":
       self.cell().shapes(self.lay_layer).insert(pya.Box(-self.w_dbu/2,-self.w_dbu/2,self.w_dbu/2,self.w_dbu/2)) #Inserts a box with the initial feature size 
   
       #Create a loop to insert the remaining elements of the array
     while r < patternwidth_dbu/2:
        r_dbu = r_dbu + spacing_dbu + w_dbu #Increment the radius to the position of the centre of the next feature in the array.
        a = 0 #Reset the value of the angle before the loop to create the array continues
        
        while a < math.pi *2:
         if self.featureshape == "Circle":
         #Compute the circle at the new location in the array
           for i in range(0,self.npoints):
             pts = [] #List to house all the points of the circle co-ordinates
             pts.append(pya.Point.from_dpoint(pya.DPoint((r_dbu + w_dbu/2) * math.cos(i * da), (r_dbu + w_dbu/2) * math.sin(i * da)))) #Adding the co-ordinates to the pts list
             self.cell.shapes(self.lay_layer).insert(pya.Polygon(pts))
         elif self.featureshape == "Square":
           self.cell().shapes(self.lay_layer).insert(pya.Box(-self.w_dbu/2+ r_dbu*math.cos(a),-self.w_dbu/2 + r_dbu*math.sin(a),self.w_dbu/2 +  r_dbu*math.cos(a), self.w_dbu/2 + r_dbu*math.sin(a))) #Inserts a box with the initial feature size at the allocated location in the array
         a = a + math.pi*2 / self.n

  # optional:
  # def can_create_from_shape_impl(self):
  #   TODO: determine if we have a shape that we can use to derive the 
  #   PCell parameters from and return true in that case
  # 
  # optional:
  # def parameters_from_shape_impl(self):
  #   TODO: change parameters using set_x to reflect the parameter for the
  #   given shape
  # 
  # optional:
  # def transformation_from_shape_impl(self):
  #   TODO: return a RBA::Trans object for the initial transformation of
  #   the instance
  
# TODO: add more PCell classes ..

# The PCell library declaration
# A PCell library must be declared by deriving a custom class from RBA::Library.
# The main purpose of this class is to provide the PCell declarations and to register itself
# with a proper name.
# TODO: change the class name
class PCellLib(pya.Library):

  def __init__(self):
  
    # TODO: change the description
    self.description = "My PCell library"
    
    # register the PCell declarations
    # TODO: change the names
    self.layout().register_pcell("PolarArray", PolarArray())
    # TODO: register more PCell declarations
    
    # register our library with the name "PCellLib"
    # TODO: change the library name
    self.register("PCellLib")
    
# instantiate and register the library
# TODO: change the library name
PCellLib()

`

klayoutmatthias, 2022-03-18:
I took the liberty to format your code properly. This was hardly readable.

Setting breakpoints inside interactive actions (which PCell code evaluation is one) is kind of unstable. In a particular condition (drag and drop of PCell library into layout) I could trace it down to a Qt bug (QDrag object gets deleted) which I could not work around. Debugging coerce_parameters_impl is even more nasty as this function gets called internally on different occasions and the debugger will heavily interfere with the application. Windows specifically seems to be susceptible to crashes then. To fix this problem, I'd need to detach the debugger and place it into a separate process - that is a bigger project however.

In your case you can debug by putting print statements into your code and checking the console. You can also switch off the debugger which avoids making the application unstable. In your case, the code had a couple of issues which triggered errors in various places, so I am assume multiple failures accumulating.

My advise is not to write all the code at once and then start debugging but to start from the sample and modifying it step by step.

Here are my fixes:

1. coerce parameters: proper checking of layer and datatype, wrong "or" for the featureshape test:

```
  def coerce_parameters_impl(self):
    # TODO: use x to access parameter x and set_x to modify it's value 
    
    if self.lay.layer != 5 and self.lay.datatype != 0:
      raise(RuntimeError("layer 5 must be used for membrane patterns!"))
    if self.w <= 0:
      raise(RuntimeError("Feature width must be larger than 0!"))
    if self.r != 0: 
      raise(RuntimeError("Initial radius must be equal to 0!"))
    if self.spacing < self.w:
      raise(RuntimeError("Feature spacing must be larger than feature width!"))
    if self.a != 0:
      raise(RuntimeError("Initial angle must be equal to 0!"))      
    if self.featureshape != "Circle" and self.featureshape != "Square":
      raise(RuntimeError("Feature shape may be a Circle or Square only!"))
    if self.patternwidth < self.w:
      raise(RuntimeError("Pattern width must be larger than feature width!"))
    if self.n > 360:
      raise(RuntimeError("Cannot exceed 360 instances in a polar array!"))
    if self.npoints < 4:
      raise(RuntimeError("At least 4 points are required for a single feature shape"))
```     

2. display_text_impl: "," a string separators? use "+"!

```
  def display_text_impl(self):
    # Provide a descriptive text for the cell
    return "Polar Array(L=" + str(self.lay) + "," + str(self.featureshape) + ",PW = " + ('%.3f' %self.patternwidth) + ",FW=" + ('%.3f' % self.w) + ",S=" + ('%.3f' %self.spacing) + ")"
```

3. produce_impl: Countless issues. Many typos, wrong indents (IMHO), so much too many shapes produced, cell object being called etc. Check yourself. Here is my version (rewritten to micron-unit "D" type objects) which I **guess** is doing what you intend:

```
  def produce_impl(self):
   
     # Insert the given shape at the origin with the desired feature width
     if self.featureshape == "Circle":
       pts = [] #List to house all the points of the circle co-ordinates
       da = math.pi * 2 / self.npoints #delta in angle between points in the circle feature    
       # compute the circle
       for i in range(0, self.npoints):
          pts.append(pya.DPoint((self.w/2) * math.cos(i * da), (self.w/2) * math.sin(i * da))) #Adding the co-ordinates to the pts list
       # create the shape
       self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
      
     elif self.featureshape == "Square":
       self.cell.shapes(self.lay_layer).insert(pya.DBox(-self.w/2,-self.w/2,self.w/2,self.w/2)) #Inserts a box with the initial feature size 
   
     # Create a loop to insert the remaining elements of the array
     r = self.r
     while r < self.patternwidth/2 - 1e-10:
        r += self.spacing + self.w #Increment the radius to the position of the centre of the next feature in the array.
        a = 0 #Reset the value of the angle before the loop to create the array continues
        
        while a < math.pi *2 - 1e-10:
         if self.featureshape == "Circle":
           # Compute the circle at the new location in the array
           pts = [] #List to house all the points of the circle co-ordinates
           for i in range(0,self.npoints):
             pts.append(pya.DPoint((r + self.w/2) * math.cos(i * da), (r + self.w/2) * math.sin(i * da))) #Adding the co-ordinates to the pts list
           self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
         elif self.featureshape == "Square":
           self.cell.shapes(self.lay_layer).insert(pya.DBox(-self.w/2+ r*math.cos(a),-self.w/2 + r*math.sin(a),self.w/2 +  r*math.cos(a), self.w/2 + r*math.sin(a))) #Inserts a box with the initial feature size at the allocated location in the array
         a += math.pi*2 / self.n
```

Matthias

JT-24, 2022-03-18:
Thanks for your reply on this. I copied the code from the Macro developer and pasted it in the text editor here and noticed that some of the code adopted the correct format and some of it did not, I tried to correct it but to no avail. Thanks for assisting with the format.

I understand the tips regarding debugging inside interactive actions. In this instance, you are indeed correct, there were a number of accumulating issues. I was intending on fixing same but kept encountering the 'crash' I initially described which was inhibiting my detection of the 'bugs' in my draft code in the initial post. 

In response to each of your fixes: 

1. I was coerce parameters to ensure the layer selected by the user was chosen to be layer 5. I am presuming a much simpler way to do this would be to remove the if statement and set it directly to the layer value desired? 
`self.lay_layer = 5` or something along these lines?

2. I started with the example PCell code here and made an error in concatenating the string.

3. In terms of the indents being incorrect, is this with respect to the PEP-8 indentation style (I am assuming this also produced the unintended infinite loop)? Also I understand your comment with regards to too many shapes being produced. My other questions with this response are:
- What did you mean with respect to the cell object being called? 
- Is there any benefit to using D-type objects as opposed to converting the parameters to database units as I had set up in the initial code?
- Why the addition of - 1e-10 on the following lines of code:
-`while r < self.patternwidth/2 - 1e-10:`
-`while a < math.pi *2 - 1e-10:`

4. The crash issue is resolved. However there is an issue with these lines:

`if self.featureshape != "Circle" or self.featureshape != "Square":`
        `raise(RuntimeError("Feature shape may be a Circle or Square only!"))`

[image]

I have printed the value to the console, it returns Circle, as expected for the provided input. Since the value of self.featureshape is 'Circle' as shown here, could this be an issue in checking the value of the variable when comparing with "Circle". I have tried to include the apostrophes in the code (i.e. "'" + "Circle" + "'") and end up with the same error.

[image]

klayoutmatthias, 2022-03-18:
1. If the layer is fixed, you maybe should not make that an parameter at all

3:
* `self.cell` is the cell object, `sell.cell()` will "call" the object. A cell cannot be called
* "D" type objects (like DBox, DPolygon ...) are expressed in micrometer units, so you can directly use micrometers rather than DBU units. For PCells where the parameters are usually micrometers this is more convenient. Most functions provide interfaces for both "D" (DBox, ...) and "non-D" (Box, ...) objects.
* Never compare floating-point values directly because of accumulating rounding errors (specifically as you add small amounts continuously). Always use some tolerance (often called "epsilon" in text books). 1e-10 is a suggestion. Numerics 101.
* `self.featureshape != "Circle" or self.featureshape != "Square"` is **always** true. That is why I canged it to *and*!

JT-24, 2022-03-21:
Thanks a lot for the help! The code runs without any issues and it generates an output, however, it appears not to show the expected pattern (a polar array of circular or square features). I tried to adjust the levels in the program to show all levels in the hierarchy and it still does not appear. Do you know why this may be occurring? 

I have included a screenshot for reference (here showing a 200 x 50 instance array of the PCell) and the full code just in case I have missed anything (though the indentation is not retained when I formatted it in this post). All of the changes discussed have been incorporated. For reference, the polar array is set at 100 micron in width so should be easily visible at the scale shown.

[image]

[image]

`import pya

`# PCell template`
`# This macro template provides the framework for a PCell library`

`# It is recommended to put PCell code into namespaces.`
`# TODO: change the module name`

`# The PCell declaration`
`# Each PCell must provide a declaration. It is recommended to use the PCell name as the class name.`
`# TODO: change the class name`
`class PolarArray(pya.PCellDeclarationHelper):`

`  def __init__(self):`

`    # Important: initialize the super class`
`    super(PolarArray, self).__init__()`

`    # declare the parameters`
`    self.param("lay", self.TypeLayer, "Layer", default = pya.LayerInfo(5, 0))`
`    self.param("s", self.TypeShape, "",)`
`    self.param("n",self.TypeInt,"Number of instances in polar array",default = 4)`
`    self.param("npoints",self.TypeInt,"Number of points in the circle", default = 64)`
`    self.param("w",self.TypeDouble,"Width/Diameter",default = 0.5)`
`    self.param("r",self.TypeDouble,"Radius",default = 0)`
`    self.param("spacing",self.TypeDouble,"Spacing",default = 0.5)`
`    self.param("a",self.TypeDouble,"Angle",default = 0)`
`    self.param("featureshape",self.TypeString,"Feature Shape",default = "Circle")`
`    self.param("patternwidth",self.TypeDouble,"Pattern Width", default = 100)`

`  def display_text_impl(self):`
`    # Provide a descriptive text for the cell`
`    return "Polar Array(L=" + str(self.lay) + "," + str(self.featureshape) + ",PW = " + ('%.3f' %self.patternwidth) + ",FW=" + ('%.3f' %self.w) + ",S=" + ('%.3f' %self.spacing) + ")"`
`  def coerce_parameters_impl(self):`
`    # TODO: use x to access parameter x and set_x to modify it's value`
`    print(self.featureshape)`
`    if self.w <= 0:`
`      raise(RuntimeError("Feature width must be larger than 0!"))`
`    if self.r != 0:`
`      raise(RuntimeError("Initial radius must be equal to 0!"))`
`    if self.spacing < self.w:`
`      raise(RuntimeError("Feature spacing must be larger than feature width!"))`
`    if self.a != 0:`
`      raise(RuntimeError("Initial angle must be equal to 0!"))`      
`    if self.featureshape != "Circle" and self.featureshape != "Square":`
`      raise(RuntimeError("Feature shape may be a Circle or Square only!"))`
`    if self.patternwidth < self.w:`
`      raise(RuntimeError("Pattern width must be larger than feature width!"))`
`    if self.n > 360:`
`      raise(RuntimeError("Cannot exceed 360 instances in a polar array!"))`
`    if self.npoints < 4:`
`      raise(RuntimeError("At least 4 points are required for a single feature shape"))`
     
`  def produce_impl(self):`
`     # TODO: produce the cell content`
`     # i.e. self.cell().shapes(self.l_layer).insert(pya.Polygon(...))`
   
`     #Insert the given shape at the origin with the desired feature width`
`     if self.featureshape == "Circle":`
`       pts = [] #List to house all the points of the circle co-ordinates`
`       da = math.pi * 2 / self.npoints #delta in angle between points in the circle feature`
`       # compute the circle`
[... 26 lines cut ...]

`# The PCell library declaration`
`# A PCell library must be declared by deriving a custom class from RBA::Library.`
`# The main purpose of this class is to provide the PCell declarations and to register itself`
`# with a proper name.`
`# TODO: change the class name`
`class PCellLib(pya.Library):`

`  def __init__(self):`
  
`    # TODO: change the description`
`    self.description = "My PCell library"`
    
`    # register the PCell declarations`
`    # TODO: change the names`
`    self.layout().register_pcell("PolarArray", PolarArray())`
`    # TODO: register more PCell declarations`
    
`    # register our library with the name "PCellLib"`
`    # TODO: change the library name
    self.register("PCellLib")`
    
`# instantiate and register the library`
`# TODO: change the library name`
`PCellLib()`

JT-24, 2022-03-21:
Thanks for the clarification, the macro is running as expected. Thank you for all of the help, I appreciate it!

closed 2022-03-31 by JT-24