KLayout/klayout #1041: A fundamental bug concerning '==' method in Python package klayout

Z-KN, 2022-03-23:
[Enrivonment]: The version of Python package klayout is 0.27.8; the version of software KLayout is 0.27.5.
[Bug description]: When comparing two objects with the '=='  method, the comparison fails to function correctly.
[Case]: Running the following code:
```
import klayout.db as db
ly = db.Layout()
l1 = ly.layer(1, 0)
ly.dbu=0.1
top_cell = ly.create_cell("TOP")

p1=db.Point(1,1)
p2=db.DPoint(1,1)
p3=db.Point(2,2)
p4=db.DPoint(2,2)

b1=db.Box(p1,p3)
b2=db.DBox(p2,p4)

top_cell.shapes(l1).insert(b1)
top_cell.shapes(l1).insert(b2)
assert (p1==p2)
assert (b1==b2)
assert (b1.area()==b2.area())

ly.write("Case.gds")
```
None of the assertions should pass, because they are not the same at all, which can easily be confirmed by looking at the graph from KLayout. However, they all pass wrongly.

[Potential cause]: (1) The author fails to take "ly.dbu" into consideration. In fact, the assertion could pass only when ly.dbu=1.
(2) The author fails to check the type of the variable on both sides of the equal sign '==' (in the API document, '==' method requires two variables of identical type, however).

[image]

klayoutmatthias, 2022-03-24:
It's a matter of definition. You are comparing different things and the result is essentially undefined.

Basically you compare a Box vs. DBox which are different objects. In the API they their use implies a certain unit, but the box itself first of all is a numerical entity.

If you insert a Box, it is taken to be in database units. So if you have a database unit of 1 nm, a box of (0, 0; 10, 10) will render a 10 x 10 nm box. That scaling is not a feature of the box, but of the layout that receives it.

When you insert a DBox with the same dimensions - i.e. (0.0, 0.0; 10.0, 10.0) it will always render a 10 x 10 µm box, independent of the database unit of the layout object. 

So numerically, these objects are identical (10 == 10 and 0 == 0), but their *rendering* is different.

Z-KN, 2022-03-25:
Yes, I agree, it is a definition issue.

But intuitively, we should judge whether two objects are 'equal' on the basis of the actual layout, rather than objects in a C++ or Python program; after all, scripts serve the layouts, do they? I think it is confusing when checking layouts and finding different patterns that were assumed identical.

All in all, it is a suggestion that the author adds type examination in the Python package for clarification and robustness of the program.

klayoutmatthias, 2022-03-25:
DBox objects have a second life as Box objects with fractional coordinates. Such objects are useful for intermediate results in computations when you do not want to loose precision. In this context, the compare of Box and DBox makes sense. Qt for example has a similar concept of QRect and QRectF. Qt also allows comparing QRect and QRectF by virtue of implicit type conversion.

My advise is: just stay within the Box or DBox world consistently (or Edge/DEdge, Polygon/DPolygon, Path/DPath etc.) to avoid unintuitive effects. It's better anyway, because you may get rounding if you mix DBox with Box.

Matthias

Z-KN, 2022-03-26:
Thank you for your advice.

closed 2022-03-26 by Z-KN