KLayout/klayout #1042: AIM PDK installation

ansingh07, 2022-03-24:
Hi,

I trying to install AIM Photonics Libraries (v4.5) in Klayout (v 0.27.8) on a windows computer. I put the AIMPhotonics_PDK in Klayout\salt\ folder and then importing new technology in Klayout using technology manager, however it's not importing any libraries somehow. I am able to see AIM photonics PDK in Technology drop down menu but it's not showing the libraries. I tried different version of Klayout but no success. Any help is appreciated.

thanks
Anshuman

klayoutmatthias, 2022-03-24:
This package is not part of this project, so I cannot help. 

I do not even see it in the package index. Is that a public Salt package?

ansingh07, 2022-03-24:
Hi,

This the pdk package from AIM foundry. I followed their instructions to
install it in Klayout but it’s not working for me. They mention to use
SiEPIC version 0.80 but I don’t know how to install the older version
through klayout package manager and I am using version 0.89.

Thanks
Anshuman

klayoutmatthias, 2022-03-25:
This is confusing. How do you install that package?

Again, I am not the author of this package. If you think it is a KLayout bug, please describe in detail how I can reproduce it.

klayoutmatthias, 2022-03-25:
I guess reproducing is not an option as the PDK is not publicly available.

bergman1978, 2022-03-26:
Ansingh07,
   Send email to bbergman@sunypoly.edu or aimphotonics.com/helpdesk. Glad to help you use our pdk.

bergman1978, 2022-03-26:
Your install actions do not match the instructions in AIM_KlayoutPDKUserGuide_ACT_TDUA_01192022.pdf. Please review that document but I will provide an important snip of details here.      "Copy the EPDA assisted tapeout\KLayout\AIMPhotonics PDK folder to the following directory.
The destination path may be different depending on your KLayout installation. If you do not
have the tech folder, please create it.
– Windows: C:\User\YOUR USER ID\KLayout\tech
– Linux:∼/.klayout/tech
Once you successfully installed SiEPIC-tools and AIMPhotonics PDK, you will see the SiEPIC-tools
showing up on KLayout’s main menu and the AIM Photonics’ technologies listed in the drop-down
menu on the toolbar. To use the PDK, please select “AIMPhotonics PDK” from the drop-down menu
and click on it. The technology of the currently selected layout is shown in the status bar located at
the lower-left corner of the main window. Please see Figure 1 for an example KLayout window with
SiEPIC-Tools and AIMPhotonics PDK installed."

ansingh07, 2022-03-28:
Hi,

I followed the instructions on the user guide first since it didn't work, I
copied the PDK folder into KLayout\Salt where other technology
libraries were present, thinking that it might work.

 Nevertheless, I created the tech folder as you mentioned and copied the
AIM PDK folder there, but still no success. I am attaching the screenshot
of my KLayout window for your reference. Have a look and let me know what
you think?

thanks
anshuman

bergman1978, 2022-03-28:
Can you call me?
Bart Bergman
952-353-0475

closed 2022-12-06 by klayoutmatthias