KLayout/klayout #1049: Strange hatching issue occurring with macro creating circular polygons

JT-24, 2022-03-31:
Hi,

I've written a macro which inserts an array of polygons (either square or circular in shape) with the capability to enhance the spacing and size, ultimately as a way of creating a 'gradient' of a particular feature over an area. At a later stage I am planning on using an angle parameter to rotate the feature but that is outside of the scope of this post. I am just noting that here as it is currently a parameter I am including in the code which is not used as of yet. 

I have noticed that the square array works just fine. However, when a circular array is inserted, there seems to be something strange occurring with the hatching (as shown below). It is quite dense and irregular compared to other circles I have generated (using almost identical code) and it is causing issues in processing time when trying to compute large arrays. I was wondering if anyone else has encountered this before? I have included a screenshot of an array of circular features with typical hatching (from other code not shown here) as well as a screenshot of an array of square features (generated from the code shown here) which does not show this hatching issue. 

I suspect the issue must lie in this section of code: 

```
if self.featureshape == "Circle":
          # Compute the circle at the new location in the array
            pts = [] #List to house all the points of the circle co-ordinates
            for i in range(0,self.npoints):
              pts.append(pya.DPoint(colp + (self.w/2)*(self.colm**j)*math.cos(i*da), rowp + (self.w/2)*(self.rowm**k)*math.sin(i*da))) #Adding the co-ordinates to the pts list
              self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
```

However it is almost identical to code I have used before, I have only added in a couple of terms to magnify the feature e.g. `self.colm**j` and `self.rowm**k`. Removing these does not influence the hatching issue. The number of points I am currently using is 64 for a circle. If you could help me find what I have missed, that would be much appreciated!

[image]

[image]

[image]

My code is as follows:

```
import pya

# PCell template
# This macro template provides the framework for a PCell library

# It is recommended to put PCell code into namespaces.
# TODO: change the module name

# The PCell declaration
# Each PCell must provide a declaration. It is recommended to use the PCell name as the class name.
# TODO: change the class name
class Gradientarray(pya.PCellDeclarationHelper):

  def __init__(self):

    # Important: initialize the super class
    super(Gradientarray, self).__init__()

    # declare the parameters
    self.param("lay", self.TypeLayer, "Layer", default = pya.LayerInfo(5, 0))
    self.param("featureshape", self.TypeString, "Feature Shape", default = "Circle")
    self.param("patternwidth",self.TypeDouble,"Pattern Width", default = 4)
    self.param("patternheight",self.TypeDouble,"Pattern Height",default = 10)
    self.param("npoints",self.TypeInt,"Number of points in the circle", default = 64)
    self.param("w",self.TypeDouble,"Width/Diameter",default = 0.5)
    self.param("fspace",self.TypeDouble,"Feature spacing",default = 1)
    self.param("fspacerowm",self.TypeDouble,"Feature spacing row mag",default = 1)
    self.param("fspacecolm",self.TypeDouble,"Feature spacing col mag",default = 1)
    self.param("rowm", self.TypeDouble, "Row Feature Mag", default = 1)
    self.param("rowa", self.TypeDouble, "Row Angle", default = 0)
    self.param("colm", self.TypeDouble, "Col Feature Mag", default = 1)
    self.param("cola", self.TypeDouble, "Col Angle", default = 0)
    
  def display_text_impl(self):
    # Provide a descriptive text for the cell
    return "Parray(L=" + str(self.lay) + str(self.featureshape) +  ",PW =" +  ('%.3f' %self.patternwidth) + ",PH =" +  ('%.3f' %self.patternheight) + ")"
    
  def coerce_parameters_impl(self):
  
    if self.w <= 0:
      raise(RuntimeError("Feature width must be larger than 0!"))
    if self.fspace < (self.w - 1e-10):
      raise(RuntimeError("Feature spacing must exceed feature width!"))
    if self.fspacerowm <= 0 or self.fspacecolm <= 0:
      raise(RuntimeError("Feature space magnification must be greater than 0!"))
   
  def produce_impl(self):
    da = math.pi * 2 / self.npoints #delta in angle between points in the circle feature    
    rowp = 0 
    colp = 0
    j = 0
    k = 0 
    while colp < (self.patternwidth - 1e-10): 
        j += 1
        k += 1
        while rowp < (self. patternheight - 1e-10):      
          if self.featureshape == "Circle":
          # Compute the circle at the new location in the array
            pts = [] #List to house all the points of the circle co-ordinates
            for i in range(0,self.npoints):
              pts.append(pya.DPoint(colp + (self.w/2)*(self.colm**j)*math.cos(i*da), rowp + (self.w/2)*(self.rowm**k)*math.sin(i*da))) #Adding the co-ordinates to the pts list
              self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
          elif self.featureshape == "Square":
              self.cell.shapes(self.lay_layer).insert(pya.DBox(colp - (self.colm**j)*self.w/2, rowp - (self.rowm**k)*self.w/2, colp + (self.colm**j)*self.w/2, rowp + (self.rowm**k)*self.w/2)) #Inserts a box with the magnified feature size at the allocated location in the array  
          rowp += self.fspace*(self.fspacerowm**k) 
        colp += self.fspace*(self.fspacecolm**j) 
        rowp = 0 
    
class MyLib(pya.Library):

  def __init__(self):
  
    # Set the description
    self.description = "My First Library"
    
    # Create the PCell declarations
    self.layout().register_pcell("Gradientarray", Gradientarray())
    # That would be the place to put in more PCells ...
    
    # Register us with the name "MyLib".
    # If a library with that name already existed, it will be replaced then.
    self.register("MyLib")

# Instantiate and register the library
MyLib()
```

klayoutmatthias, 2022-04-02:
It's not a hatching issue. Your code is wrong.

```
            for i in range(0,self.npoints):
              pts.append(pya.DPoint(colp + (self.w/2)*(self.colm**j)*math.cos(i*da), rowp + (self.w/2)*(self.rowm**k)*math.sin(i*da))) #Adding the co-ordinates to the pts list
              self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
```

should be

```
            for i in range(0,self.npoints):
              pts.append(pya.DPoint(colp + (self.w/2)*(self.colm**j)*math.cos(i*da), rowp + (self.w/2)*(self.rowm**k)*math.sin(i*da))) #Adding the co-ordinates to the pts list
            self.cell.shapes(self.lay_layer).insert(pya.DPolygon(pts))
```

(observe indention in Python!)

Matthias

JT-24, 2022-04-04:
Noted, thanks for the feedback.

closed 2022-04-04 by JT-24