KLayout/klayout #1051: Importing net names from gerber X2

jfdbx, 2022-04-02:
gerber X2 files can contain net names. It would be great to use them for tracing nets rather than manually adding labels to do so.
Thanks for KLayout, a great tool.
[Tutorial5.zip](https://github.com/KLayout/klayout/files/8402617/Tutorial5.zip)

klayoutmatthias, 2022-04-04:
I have translated the sample from Tutorial5.zip with the first version from the issue-1051 branch. Please see attached file. Will this be useful?

[Tutorial5_with_labels.zip](https://github.com/KLayout/klayout/files/8413520/Tutorial5_with_labels.zip)

For the implementation I had to assume that the "%TO.." sections separate regions of positive polygons. Specifically, each "%TO.." statement triggers a flush of the graphics queue, so it cannot be in the middle of a sequence of clear polygons for example (I reason that does not make much sense anyway). A single representative text is generated for each "%TO.." statement and attached to one of the polygons or paths generated by the following pieces.

Matthias

klayoutmatthias, 2022-04-04:
BTW: I cross-checked with the reference X2 viewer (https://gerber-viewer.ucamco.com/) and your test case, but it does not seem to properly attach nets to the SMD pads. They are all /GND.

jfdbx, 2022-04-05:
> Hi Matthias,
>
> Looks great. I will trace the nets on that and let you know but it should work as the labels are located on the right layers at the right positions.
>
> How did you do that ? Did you have to change something in the code or is it through macros ?
>
> I am struggling to get a sample from Altium. It looks like they generate X2 but without labels… I am asking to Altium support.
>
> Regards
> Jean-François

I traced the nets : It’s perfect.

Here is what it look like once imported in my 3D Field Solver:

***@***.***

KR
Jean-François

closed 2022-04-09 by klayoutmatthias

jfdbx, 2022-10-11:
Hi Matthias,

Looks great. I will trace the nets on that and let you know but it should work as the labels are located on the right layers at the right positions.

How did you do that ? Did you have to change something in the code or is it through macros ?

I am struggling to get a sample from Altium. It looks like they generate X2 but without labels… I am asking to Altium support.

Regards
Jean-François