KLayout/klayout #1063: Does method layout.read always return layouts involving shapes of simple_polygon rather than polygon?

Z-KN, 2022-04-17:
Firstly I create a "db.Polygon" with holes and execute layout.write("xxx.gds"), then load it through "layout.read("xxx.gds"). When I analyze the shape in the loaded layout, the returned shape will always be "db.SimplePolygon" with holes resolved. Also, the KLayout viewer will display a simple polygon in "Polygon Properties". It seems the reading method automatically converts a complex polygon to a simple polygon. However, I want the raw complex polygon with holes instead of a simple polygon, which sometimes has edges overlapped.

What should I do?

ejprinz, 2022-04-17:
I think GDSII cannot store polygons with holes. Try saving in OASIS format and see whether this works.

Z-KN, 2022-04-18:
Thank you, I agree. The user should pre-process the layout or turn to another format.

closed 2022-04-18 by Z-KN