KLayout/klayout #1064: Console does not show in macro editor

jamesmeiklejohn, 2022-04-19:
On Windows 11 with a new install of KLayout 0.27.8, the console does not show below the editor window.

[image]

Kazzz-S, 2022-04-19:
I'm not a Windows 11 user, but...

[image]

jamesmeiklejohn, 2022-04-19:
I tried that as well as adjusting the size of the window and it doesn't show - also tried with a different monitor resolution.

klayoutmatthias, 2022-04-22:
You can try to clear the configuration cache. Maybe there is something kept there which blocks the readjustment.

To clear the configuration, remove or rename the file called "klayoutrc" inside KLayout's home folder. On Windows you will usually find this file in c:\users\youraccountname\KLayout. This should restore the original settings and clear any stored display configurations.

Matthias

jamesmeiklejohn, 2022-04-24:
thanks this worked! Possibly upgrading to Windows 11 is the origin of this if the cache was generated on 10?

klayoutmatthias, 2022-04-24:
Possibly. This specific configuration cache is managed by Qt and it is hard to debug this issue. 

Matthias

closed 2022-04-24 by klayoutmatthias

NemoAndrea, 2024-02-28:
> I'm not a Windows 11 user, but...
> 
> [image]

For me removing the `klayoutrc` file did not solve the issue (Windows 10), but from the image above I realised I could drag the console into view from the right side of the text editor window. So instead of below the text editor (as in quoted image), from the right side. 

Adding in case other people have the same issue.