KLayout/klayout #1065: VIA:SIZE is order dependent

maliberty, 2022-04-19:
When using a map file to load LEF/DEF and a VIA:SIZE is given, the tool appears to require a particular order of dimensions.  Other tools do not have this requirement and I think KLayout should not either.

For an example, I took testdata/lefdef/viasize and hacked up two map files where the only different in the order to the dimensions:
```
< VIA1 VIA:SIZE:0.02x0.024 8 2
---
> VIA1 VIA:SIZE:0.024x0.02 8 2
```

If I import I get different results for the bar cut shape:

Ok:
[image]

Fail:
[image]

maliberty, 2022-04-19:
[klayout-1065.zip](https://github.com/KLayout/klayout/files/8516354/klayout-1065.zip)

klayoutmatthias, 2022-04-22:
Thanks for this hint and the test case. I actually wasn't sure whether "tall" and "flat" vias are considered different.

I have changed the code such that the via size is normalized to "tall" (so the layer names reflect the tall version) and in the map file you can specify any of the two versions - both via types will match then.

I have included this fix in the "lefdef-enhancements" branch, but will merge that to master and release it in 0.27.9 soon.

Best regards,

Matthias

closed 2022-04-23 by klayoutmatthias