KLayout/klayout #1066: Why this project named KLayout

YifeiCN, 2022-04-20:
I'm curious about the meaning of K :)

mi-and-core, 2022-04-22:
One of many interpretations: K stands for the first letter of Matthias' last name

klayoutmatthias, 2022-04-22:
Let me tell you it's not for "KDE" :)

The short story is that I published this project under the name "Microscope" initially, but then I learned that this is a registered trademark for a CAD file viewer. So I had to change it quickly and "KLayout" simply was the first thing that came to my mind. 

You can take a look at the archived old documentation here: https://www.klayout.de/0.16/rba_index.html - the background image of the left panel still shows a microscope:

[image]
 
Matthias

YifeiCN, 2022-04-23:
> 

I didn't expect to get your reply in person. Thank you very much for your reply. At first, I thought KLayout meant the king of layout ; )

closed 2022-12-06 by klayoutmatthias