KLayout/klayout #1067: State of Toolbar in context menu and View | Toolbar are not always synchronized

EugeneZelenko, 2022-04-26:
When Toolbar is hidden via Toolbar in context menu, check mark is still in View | Toolbar menu. This problem exists in 0.27.9 and 0.26.4.

klayoutmatthias, 2022-04-26:
You mean this checkbox is still set?

[image]

In my case it isn't (Ubuntu-20). What is the system you're on?

Matthias

EugeneZelenko, 2022-04-26:
Yes. I use Ubuntu 18, your build for Ubuntu 18 (0.27.9) and custom build on Red Hat (0.26.4).

klayoutmatthias, 2022-04-28:
I cannot see this issue on Windows too.

Maybe it is related to the Window Manager. What is the manager you are using?

Matthias

EugeneZelenko, 2022-04-28:
Looks like it's GNOME:

`XDG_CURRENT_DESKTOP=ubuntu:GNOME`
`GDMSESSION=ubuntu`

Eugene.

klayoutmatthias, 2022-12-06:
I'm having trouble seeing that issue. Is that still valid?

EugeneZelenko, 2022-12-06:
Yes, problem is still there. I also tried 0.27.13.