KLayout/klayout #1075: Edit layer specification doesn't work well with PCells

jurask, 2022-05-06:
When there is PCell inserted in layer, changing layer properties (number, datatype) does not work properly. When the GDS file is saved, PCell geometry is still in original layer.

Steps to reproduce:
1. create new layout
2. Insert some PCell into it
3. Change layer number via Edit layer specification - PCell geometry will be under new layer number
4. Save the GDS
5. Restart KLayout and open the file
6. PCell geometry will be in the original layer as before point 3

klayoutmatthias, 2022-05-06:
I see.

The problem is that you're renaming the layer which temporarily shifts the PCell's shapes to the new layer. But the PCell parameter is still referring to the old layer. So when the PCell is regenerated on loading, the old layer will be restored.

I think it's possible to make the layout rename change the PCell parameters as well.

Thanks for reporting this.

Matthias

closed 2022-06-18 by klayoutmatthias

jurask, 2022-07-04:
Thanks for help, when I edit properties from GUI, it works now fine. Unfortunately, when I edit the layer from script using Layout.set_info(index, pya.LayerInfo(n, datatype)) the PCells geometry is still under original layer number. Should I open new issue, or am I doing something wrong?

klayoutmatthias, 2022-07-04:
No, this will not work. PCell parameters and the layer info data fields are separate entities. If you do a scripted change, you'll need to modify the PCell parameters.