KLayout/klayout #1092: improve python API and documentation

joamatab, 2022-06-06:
How can we improve the python API?

I love klayout and I would like to use it more.
The main issue i ran is into is that all my workflow is in python and that I can't find the python function to the things I want to do in klayout

it would be great to have it as documented as the one in [gdspy](https://github.com/heitzmann/gdspy) or [gdstk](https://github.com/heitzmann/gdstk)

@klayoutmatthias 
@thomaslima 
@proppy

joamatab, 2022-06-07:
I found this tutorial helpful
https://www.klayout.org/klayout-pypi/

How could we have a full API documentation for the python API? SImilar to the [gdstk one](https://heitzmann.github.io/gdstk/reference_python.html#) 

@heitzmann

klayoutmatthias, 2022-06-07:
There is a full API documentation here: https://www.klayout.de/doc-qt5/klayout.pdf (PDF) or https://www.klayout.de/doc-qt5/code/index.html (reference) and here https://www.klayout.de/doc-qt5/programming/index.html (entry)

After all, the API is pretty complex (I am counting 211 classes not including the Qt ones) and bilingual. I'm not a writer and documentation is a hardly valued but time consuming job.

klayoutmatthias, 2022-06-07:
I feel that the forum would be rather helpful if Vanilla did no have such a lousy search feature. 

Does anyone know a better system with the option to import a Vanilla forum database?

Matthias

thomaslima, 2022-07-25:
@klayoutmatthias @joamatab 

Checkout my demo in #1125 . Honestly the current API documentation Matthias compiled is excellent for me. My only pain point was the lack of autocomplete, or any sort of type hinting or docstring lookup on my IDE (vscode). 

I have made some tweaks to the doc strings of the python module's c-extension objects to allow for [mypy's stubgen program](https://mypy.readthedocs.io/en/stable/stubgen.html) to generate 99% of the hardworking in creating these stubs. I think for all practical purposes klayout's python module feels and behaves much like a pure python package, except perhaps the error/exception handling.

klayoutmatthias, 2022-07-25:
Thanks a lot for this encouraging feedback :)

I wonder how I can improve the docstrings or give other hints (in C maybe?) to facilitate this use case. Could you provide some hints for me? And maybe also for error and exception handling - how can I improve here?

Thanks and best regards,

Matthias

klayoutmatthias, 2022-07-25:
I just noticed your pull request. Many thanks! :)

thomaslima, 2022-07-25:
> Thanks a lot for this encouraging feedback :)

You're welcome! Sometimes silence is the best feedback. That means it's working.
> I wonder how I can improve the docstrings or give other hints (in C maybe?) to facilitate this use case. Could you provide some hints for me? And maybe also for error and exception handling - how can I improve here?

You've already found my PR #1125 . Here's what I did: I had to change the way the docstrings were produced a little bit. stubgen actually looks into the documentation to find function signatures of the format `method(arg: arg_type, ...) -> ret_type` within the code. Multiple occurrences of the signature would capture overload possibilities. I also did a little trick to add `------\n` between different overload docstrings. So I had to modify mypy's stubgen to look for those strings and split the docstring accordingly (turns out they have a few bugs in their docparser). The end result is what you see. I have made no attempt to parse the docstring and remove the `@brief` and `@param` as you did in the layGSI parser. I think it's good enough since I assume people who look for docstrings are very advanced users and don't mind the extra keywords.

Regarding the exception handling, the issue I had over the years was explaining the content of obscure RuntimeError messages to others. When they run a script that throws an exception from within klayout, its cause might be a bit opaque at times. From memory, most of the errors came from accidentally giving variables to arguments with the wrong type. When that happens, the error that comes back is usually not "arg x is of the wrong type", it's usually some fatal c++ error of trying to iterate over an integer instead of a list of points for example. When the PR is merged, I think the probability will be significantly decreased if people actually use static checking on python. At this point it would be the user's responsibility to use the right signature with the tools at hand. They can also check their program with a static type checker like mypy and catch a lot of bugs like in any statically typed language.

thomaslima, 2022-08-29:
The original request can be fulfilled by reading the documentation here: https://www.klayout.de/doc-qt5/klayout.pdf (PDF) or https://www.klayout.de/doc-qt5/code/index.html (reference) and here https://www.klayout.de/doc-qt5/programming/index.html (entry) - both are listed above.

In addition, in the next release of 0.27, IDEs like VSCode will have code highlighting as shown in #1125.

closed 2022-08-29 by thomaslima