KLayout/klayout #1093: "klayoutrc" file sometimes causes runtime error

dsus4wang, 2022-06-07:
I use Klayout 0.27.9 on the newest MacOS(M1 chip).
Occasionally when I debug the macro, I changed the parameters, but the script still runs with the old parameters.
I found if I delete the <edit-inst-pcell-parameters>  in klayoutrc file,  the program would back to normal.
Could you help me find the problem?thanks.

klayoutmatthias, 2022-06-07:
I'm afraid I don't understand. 

What are the parameters you change and what does the last message mean?

Matthias

dsus4wang, 2022-06-11:
I use Klayout 0.27.9 on the newest MacOS(M1 chip).
Occasionally when I debug the macro, I changed the parameters, but the script still runs with the old parameters.
I found if I delete the edit-inst-pcell-parameters  in  klayoutrc file, the program would back to normal.
Could you help me find the problem?thanks.

klayoutmatthias, 2022-06-18:
KLayout stores the most recently used parameters. So when you instantiate a PCell again, it will receive the parameters you have used most recently. That is an intended behaviour and is for the convenience of the users who do not have to re-enter the same values.

So when you change the parameters (I assume you change the definition), then you will see the old values, that is true, but not a bug. Either you change the name of the parameter when you change the type for example, or you need to do the reset as you have described it.

Matthias

closed 2022-12-06 by klayoutmatthias