KLayout/klayout #1094: Mac: Fill Tool missing most parameters

demisjohn, 2022-06-09:
It's my first time using the Edit>Utilities>`Fill Tool` in KLayout, so it's entirely possible I'm just mistaken here.

According to [this video](https://www.youtube.com/watch?v=uP8QQ_Ggnkg), the `Fill Tool` should pop up with options for "Fill Cell" (which cell to perform filling with) like this:
[image]

However, on the Mac version of Klayout (originally tried 0.27.4, then upgraded to 0.27.10 - no difference), I am missing most of the required parameters:
[image]

If I hit [OK] I get "Fill cell not found:" indicating an empty parameter sent to the function.
[image]

Perhaps there's a bug in the MacOS version causing these fields to be missing?

### Versions
MacOS 12.2.1 (21D62)
KLayout 0.27.10
[image]

Kazzz-S, 2022-06-09:
I have not used `Edit>Utilities>Fill Tool`. So I cannot instruct how to use it.

However, I think there was a change in the dialog GUI design from 0.26.x to 0.27.x.
The video tutorial uses 0.26.4.

My macOS version is 12.4 (Monterey).
Therefore, I cannot run the KLayout bundled with Python 2.7.18 you are using.
So, I've tested with the nearest one.

Please compare the images below.
In my observation, all parameter fields (and more) are distributed in four tabs in the latest version.

## 0.26.4 ##
[image]
[image]

## 0.27.10 ##
[image]
[image]
[image]
[image]
[image]

demisjohn, 2022-06-10:
Ah I see - I have to fill out fields in ALL FOUR tabs - yes indeed this works. It seemed like each tab was a different function, rather than being parameters of the same function. 

This works as expected then!

closed 2022-06-10 by demisjohn