KLayout/klayout #1096: Internal error triggered by sky130 DRC script

pepijndevos, 2022-06-14:
```
'width': Internal error: ../../../src/db/db/dbHierNetworkProcessor.cc:2095 id_new != 0 was not true in Region::width_check in Executable::execute
  /home/pepijn/.klayout/drc/sky130B.lydrc:454:in `execute'
  :/built-in-macros/drc_interpreters.lym:27:in `instance_eval'
  :/built-in-macros/drc_interpreters.lym:27:in `execute'
```
Layout:
[inverter.zip](https://github.com/KLayout/klayout/files/8900497/inverter.zip)

DRC script:
https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/klayout/sky130.lydrc

Let me know any additional info needed.

klayoutmatthias, 2022-06-14:
Thanks for this report.

My first attempt to reproduce the problem failed however (Ubuntu 20).

I used the "inverter" layout and simply ran the DRC script without further customization and ended up with a clean run. Did you use some special options/settings/inputs?

Thanks,

Matthias

klayoutmatthias, 2022-06-14:
I noticed that the layout has additional top cells, but I doubt that is important here. 

I just ran the sample on a clean ubuntu:20.04 Docker instance and the official 0.27.10 .deb image from my web site. I had to patch sky_130.lydrc to support specification of a top cell:

```
# optional for a batch launch :   klayout -b -rd input=my_layout.gds -rd report=sky130_drc.txt -r drc_sky130.drc
if $input &amp;&amp; $top
  source($input, $top)
elsif $input
  source($input)
end
```

With this I ran the DRC using:

```
klayout -b -rd input=inverter -rd top=TOP -r sky130.lydrc 
```

Still no fail :(

Matthias

pepijndevos, 2022-06-15:
Maybe it *is* important? The error started happening after I added the via pcell. What is the "correct" way to not have multiple top cells?

Meanwhile I'm trying to build from source. I'm on Arch linux, so probably a number of dependencies are a bit newer than Ubuntu.

pepijndevos, 2022-06-15:
I only see one top level cell??
[image]

Building from source it still crashes when running DRC from the GUI, but applying your patch to the drc file and running it in batch mode there is no error.

So my naive interpretation is either there *is* a problem with multiple top cells, except I only see one top cell so I'm not sure what you're talking about. Or there is a difference between the GUI and the command line.

klayoutmatthias, 2022-06-15:
You seem to have a library installed which I am lacking. So the library cells get resolved in your case and don't in mine. This leads to "defunc" lib cells, but as they still carry the original layout, this should not make a difference. In batch mode, the libraries are also not available, so I assume that the difference is between your case with the library and mine without. 

Interesting, that this leads to top cells in my case. Should not happen. I'd call this a bug or the problems are related. I'll try to fix that and get back to you.

Here is my hierarchy:

[image]

klayoutmatthias, 2022-06-15:
I was able to reproduce the problem after using the PDK from here: https://github.com/mabrains/sky130_klayout_pdk

But the layout with the library behaves weird. I cannot export it to GDS properly with skipping the PCell/Lib information. So I think there is something wrong internally. I guess I need to debug that application, but maybe also the sky130 PCell code.

Thanks for reporting the issue though.

Matthias

klayoutmatthias, 2022-06-16:
I was able to debug the problem.

Here is what happens: the device library generates zero-dimension arrays which do not throw an exception and are displayed properly, but they screw up the stream writers and the DRC engine.

Without doubt KLayout could be smarter and dismiss or normalize such instances (https://github.com/KLayout/klayout/issues/1098). As a workaround I have proposed a patch for the device lib (https://github.com/mabrains/sky130_klayout_pdk/pull/6) which avoids this problem at least for the pmos18 device by using an array dimension of at least 1. Others may still show this issue.

With this patch, the DRC executes successfully with on error (licon space violation).

Thanks for bringing up this topic. I'll close this issue and proceed along the path sketched in https://github.com/KLayout/klayout/issues/1098.

Matthias

closed 2022-06-16 by klayoutmatthias