KLayout/klayout #1105: Offer Docker images for building from source

nikosavola, 2022-06-28:
I'm having problems compiling from source on Windows, so I was wondering if one could offer Docker images for building KLayout. This could be done for e.g. Ubuntu and Windows as these are likely the most used OS.

Docker supports Windows containers so this should be possible.
Additionally, the Ubuntu containers could be used to build KLayout in the free GitHub Actions. [Windows containers are not supported yet](https://github.com/actions/runner/issues/904), but a Windows build workflow is probably possible as with actions like https://github.com/jurplel/install-qt-action.

klayoutmatthias, 2022-06-28:
My Dockerfile for building KLayout is this:

Ubuntu 22:

```
FROM ubuntu:22.04

RUN apt-get update && \
  apt-get -y upgrade && \
  export DEBIAN_FRONTEND=noninteractive DEBCONF_NONINTERACTIVE_SEEN=true && \
  echo "tzdata tzdata/Areas select Europe" >preseed.txt && \
  echo "tzdata tzdata/Zones/Europe select Berlin" >>preseed.txt && \
  debconf-set-selections preseed.txt && \
  rm preseed.txt && \
  apt-get install -y \
      gcc \
      g++ \
      make \
      libz-dev \
      qtbase5-dev \
      qttools5-dev \
      libqt5xmlpatterns5-dev \
      qtmultimedia5-dev \
      libqt5multimediawidgets5 \
      libqt5svg5-dev \
      ruby \
      ruby-dev \
      python3 \
      python3-dev \
      wget \
      openjdk-8-jdk \
      git \
      fakeroot \
      lintian \
      openssh-client \
      openssh-server \
      xvfb \
      sudo
```

CentOS8:

```
FROM centos:8

RUN sed -i 's/mirrorlist/#mirrorlist/g' /etc/yum.repos.d/CentOS-Linux-* &&\
    sed -i 's|#baseurl=http://mirror.centos.org|baseurl=http://vault.centos.org|g' /etc/yum.repos.d/CentOS-Linux-*

RUN yum -y update \
 && yum clean all \
 && yum -y install \
      gcc \
      gcc-c++ \
      make \
      qt5-qtbase \
      qt5-qtbase-devel \
      qt5-qtmultimedia \
      qt5-qtmultimedia-devel \
      qt5-qtxmlpatterns \
      qt5-qtxmlpatterns-devel \
      qt5-qtsvg \
      qt5-qtsvg-devel \
      qt5-qttools \
      qt5-qttools-devel \
      zlib-devel \
      ruby \
      ruby-devel \
      python3 \
      python3-devel \
      wget \
      fontconfig \
      gtk2 \
      dejavu-sans-mono-fonts \
      dejavu-sans-fonts \
      dejavu-serif-fonts \
      dejavu-fonts-common \
      openssh-server \
      java-1.8.0-openjdk.x86_64 \
      git \ 
      which \
      rpm-build \
      xorg-x11-server-Xvfb \
      mesa-dri-drivers \
      sudo
```

I don't believe in Windows Containers for everyone given Microsoft's license policy. Maybe that's an option for enterprise software development, but I'm lacking the $$$ revenue here. So don't mind it I'll skip that.

I'm using MSYS2 for building the Window binaries. Given how easy it is to install it, that's a fair compromise IMHO. There are basically other options (MSVC or Anaconda) if you like to explore them. Build recipes and MSYS2 setup is found here: https://www.klayout.de/build.html (look for "Building KLayout on Windows with MSYS2").

Matthias

klayoutmatthias, 2022-06-28:
Another remark: building the whole package is very resource greedy. Plus it needs the full Qt system which raises licensing questions. I don't want to put the full package into GitHub actions.

closed 2022-12-06 by klayoutmatthias