KLayout/klayout #1108: External DLL file errors after MSYS2 compilation

nikosavola, 2022-06-29:
Hello Matthias,

I'm confused on how one should get the external DLLs when building.
I'm getting the following errors for some Python, Ruby and Qt DLLs after compiling with MSYS2 on Windows
[image].
Should the building automatically copy these and something failed, or did I miss some part of the process. I ran `build.sh` in MSYS2 MinGW x64.

klayoutmatthias, 2022-07-01:
I think you are trying to run the .exe outside the MSYS shell I suppose. In this case you either need to set the PATH to include the place of the libpython.3.9.dll or copy .exe and all required DLLs into one place. That is how building on Windows is like. 

The easiest is to run KLayout from the MSYS2 prompt.

Matthias

nikosavola, 2022-07-04:
> I think you are trying to run the .exe outside the MSYS shell I suppose. In this case you either need to set the PATH to include the place of the libpython.3.9.dll or copy .exe and all required DLLs [...]

Ah, thank you. I didn't know it had to be run from there.

closed 2022-07-04 by nikosavola