KLayout/klayout #1113: Palettes don't respect Windows scaling

nikosavola, 2022-07-05:
Currently color palettes and stipples in the _Layers_ tabs don't respect Windows scaling. The text for the layer names does, leaving the examples a bit smaller.

Additionally, the Palettes (Color, Stipple, Line Style) in the settings don't follow Windows scaling either, resulting in the following between two monitors with different scaling

[image]

[image]

Maybe similar to #94 

A solution might exist by setting a reference DPI and scaling that https://stackoverflow.com/a/60318784

klayoutmatthias, 2022-07-08:
I don't own high-DPI hardware, so it's difficult for me to reproduce the issue. But I'll try.

klayoutmatthias, 2022-12-06:
0.28 will fix this issue.

closed 2022-12-06 by klayoutmatthias