KLayout/klayout #1114: Failed to run klayout.exe

kongfu1994, 2022-07-07:
Compiled according to the steps, run Klayout.exe, can not open without any error

klayoutmatthias, 2022-07-08:
Please provide more information about your system, the build environment, version etc.

nikosavola, 2022-07-08:
> Compiled according to the steps, run Klayout.exe, can not open without any error

I recently had problems with missing DLL-files in https://github.com/KLayout/klayout/issues/1108.
It is easiest to open _klayout.exe_ from the MSYS terminal in this case.

klayoutmatthias, 2022-07-11:
Well, building, running and debugging binaries takes some knowledge. I can't help with that - you need to be familiar with how DLLs are managed and how to debug DLL problems ([Dependencies is my friend](https://github.com/lucasg/Dependencies)).

In the sources you will find a script (`scripts/deploy-win-mingw.sh`) which does packaging for you - provided you have installed NSIS for creating the installer. This script needs to be run from MSYS2 prompt and is finally the one I use to create the binary distributions with.

As @kongfu1994 has not yet explained what his actual problem is, I tend to close this issue.

kongfu1994, 2022-07-13:
[image]

I use MSYS2 to build klayout,and install the packages with pacman:pacman -S make zip mingw-w64-x86_64-gcc mingw-w64-x86_64-make mingw-w64-x86_64-python3 mingw-w64-x86_64-qt5 mingw-w64-x86_64-ruby

and run this script ./scripts/deploy-win-mingw.sh -64

why cannot run compiler 'g++'

klayoutmatthias, 2022-07-13:
Please read the documentation: Either you proceed like on Linux with "build.sh" *or* you use "./scripts/deploy-win-mingw.sh". In the latter case, use a standard MSYS2 shell, *not* MSYS MinGW. You will also need NSIS installed and it may be required to patch the build script to point it to NSIS.

I recomment build.sh. 

I just checked myself and compilation passed with "build.sh". However, I found that with g++ 12.1 from current MSYS there is a problem with the instantiation of certain objects and klayout.exe will not start. I do not know yet if that is a compiler issue and whether there is a workaround.

klayoutmatthias, 2022-07-14:
Update: the problem isn't a KLayout issue but appears to be related to `std::type_info::before` which does does not indicate some type info structures as different which in fact they are. Both type infos come from different DLL's, so I assume the handling of such type infos inside gcc 12.1/MinGW is broken.

So bottom line is: for building KLayout MSYS2 with gcc 12.1 KLayout needs a patch.

kongfu1994, 2022-07-15:
[image]

Thanks for the reply. Now, I have successfully compiled version 0.28 and version 0.27.10 with msys2, but there is a common problem. After double-clicking klayout.exe, there is no response. In the task manager, the Klayout Application flashed by.

kongfu1994, 2022-07-16:
Please let me know when the patch is released, thanks

klayoutmatthias, 2022-07-17:
I have published a patch which makes master and 0.27 branches build on latest MSYS2. I hope this will stay so for some more rolling MSYS updates. 

I have also updated the instructions: https://www.klayout.de/build.html#build_windows_msys

Please note the part about creating a file defining the Python paths.

Matthias

kongfu1994, 2022-07-18:
Thanks for the support, klayout can run normally based on the latest code

closed 2022-07-18 by kongfu1994