KLayout/klayout #1116: Add "Apply" button to "Instance Properties" dialog

Partmedia, 2022-07-09:
This is a feature request to add an "Apply" button to the "Instance Properties" dialog (see image below).

Sometimes, I want to change the PCell parameters and see how it changes the layout, without immediately closing the dialog. Adding an "Apply" button would let me do that.

[image]

klayoutmatthias, 2022-07-11:
The parameters should be applied automatically when you change the values. 

However you need to leave the entry field before the parameters get read and transferred into the instance. You can do so by pressing "Tab" for example or by selecting a different entry field with the mouse (change of focus).

Maybe you hit "Enter" and (unintentionally) trigger "Ok" while doing so? I can stop "Ok" from being triggered by "Enter" if that helps, so "Enter" would basically do something like "Apply".

Partmedia, 2022-07-11:
Thank you for taking a look at this. Now that I know that the parameters are applied automatically, I wonder how I originally missed this!

I think what happened was that the dialog popped up and covered the cell that I was editing, and so when I updated the parameters, I couldn't see the cell update automatically until I dismissed the dialog.

Looking at other software, it seems like hitting ENTER to trigger OK and close the dialog is pretty standard behavior, so perhaps you should just keep that.

Perhaps the answer is to add a little text label near the OK button saying that changes are applied automatically? Now that I know about this behavior, I now think that it's fine as is...

nikosavola, 2022-07-12:
This brings to mind that some large PCells might take a few seconds to update. In this case if one is updating many parameters, it might be nice to have an option to not auto-update but require pressing apply. On the other had the current way is very useful and cluttering the dialog is not ideal.

klayoutmatthias, 2022-12-06:
0.28 will allow selecting of the objects you want to change. Apply is still not there (instant update, I like this better), but you have control over what you change.

closed 2022-12-06 by klayoutmatthias