KLayout/klayout #1120: need restrictions for uninstall.exe

Bismooth, 2022-07-16:
Under windows 10 environment, after wrongly installing klayout to a root directory I want to uninstall it and reinstall in a new folder. 
So I used the uninstall.exe, after realizing that it has been taking much longer time to finish deleting files than installing them I find that the program's been deleting all folders in the root directory by alphabet order. 

luckily I didn't lose too many data and after all this is a personal mistake that I should take responsibility. but similar disaster may occur to anyone who is trying to install the software so I wonder if some restrictions could be added to the uninstall program, avoiding it from exterminating everything in its directory. Another solution may be automatically create a folder for the software in install procedure so the software won't be abruptly installed in a root directory by mistake or carelessly.

klayoutmatthias, 2022-07-16:
I frequently use Uninstall from the Software control panel without issues. This is the normal way of uninstalling. I have not heard of anyone using "uninstall.exe".

"uninstall.exe" is provided through the NSIS installer system and changing that is beyond my abilities.

Bismooth, 2022-07-17:
thank you for your reply and I'll use the contorl panel more next time.
and I guess it is ok to close this issue.

closed 2022-07-17 by Bismooth

stefanottili, 2022-07-17:
How to avoid the root cause of the issue, installing in root ? Any suggestions ?

Bismooth, 2022-07-22:
install the software under a newly created folder would avoid this problem.

klayoutmatthias, 2022-07-22:
Default mode is installation in a specific folder.

Honestly I don't know if using the control panel had prevented the problem. I assume that "uninstall.exe" is added by NSIS simply to remove the installation folder entirely which normally is a perfect behaviour. I think this is also called when you run from the control panel.

But as I said, I cannot change that and I do not want to spend the effort to build my own installer. If needed, this problem should be addressed to the NSIS people.

I need to add that KLayout can be installed (in fact, is installed by default) individually for the user and within the home folder of the user. So you do not need admin permissions to install KLayout. So any user without elevated rights can do so - which I consider something special, given that most apps force you to run the installer in with elevated privileges. So if you do that and given your user does not have permissions to wipe your system, you should be safer. 

Best regards,

Matthias