KLayout/klayout #1124: m1 2.5D issues

joamatab, 2022-07-22:
How can we fix the 2.5D view on MacOS  M1 (apple silicon)?

[image]
1

Kazzz-S, 2022-07-23:
Intel Mac, too.

[image]

It looks related to [this post](https://stackoverflow.com/questions/44430594/mac-opengl-shader-error-version-150-is-not-supported). But no idea how to fix it.

klayoutmatthias, 2022-07-23:
Me neither, sorry. 

I'm not a Mac user myself (and I don't want to become one) and I cannot debug the problem.

Looks like the OpenGL implementation MacOS is either non-existing or broken. Maybe someone on a more advanced knowledge level that me may be able to help.

Matthias

klayoutmatthias, 2022-07-25:
I need to add that I can debug, but only on a VirtualBox running Catalina. I have tried to figure out how to get Qt/OpenGL running (with shader programs) on this combination without much success. I also managed to allocate one remote M1 instance on a cloud service, but was not able to repeat this experience yet. M1 cloud instance providers (e.g. ScaleWay) were solid booked out the last time I tried.

Matthias

dsus4wang, 2022-08-03:
Now ScaleWay's M1 instance is available ,could you please have a try?Thanks.
https://console.scaleway.com/asaas/servers/create

Kazzz-S, 2024-03-29:
[This forum post](https://www.klayout.de/forum/discussion/2481/opengl-error-when-running-2-5d-script) reminded me of this ticket and #941.

While investigating, I found the `#version 150` statements are embedded (hardcoded) in `src/plugins/tools/view_25d/lay_plugin/layD25ViewWidget.cc`.

In MacOS, the OpenGL version seems to be 2.1 (=> `#version 120`), while KLayout's 2.5d feature needs `#version 150` (=> OpenGL 3.2). 

I did some experiments by modifying `src/plugins/tools/view_25d/lay_plugin/layD25ViewWidget.cc`, where I forcibly set `#version 120`. Modified code ==> [layD25ViewWidget.cc.zip](https://github.com/KLayout/klayout/files/14809139/layD25ViewWidget.cc.zip)

#### Linux Mint 20 ####
```
kazzz-s@Mint20:~ (1)$ glxinfo | grep "OpenGL version"
OpenGL version string: 3.3 (Compatibility Profile) Mesa 21.2.6
```
* This version corresponds to `#version 330`.
[image]

* Two global functions appear unsupported in `#version 120`.

#### macOS Monterey ####
[image]

* The original ERROR message disappeared.
* But there is no detailed info.

Regards,
Kazzz-S