KLayout/klayout #1129: Installing klayout on debian 12 bookworm

bgarbin, 2022-07-28:
I started from: https://www.klayout.de/build.html , trying to install: Ubuntu-22/klayout_0.27.10-1_amd64.deb
  This failed as:
`
   dpkg: dependency problems prevent configuration of klayout:
   klayout depends on libc6 (>= 2.35); however:
   Version of libc6:amd64 on system is 2.33-8.
   However, the most recent version of libc6 on debian to date is 2.34.
`
I then tried installing an earlier built versions that require libpython3.8 that proves very hard to install.

Finally I am trying to build from the source and encounter issues at the beginning of the install, running:
`./builld.sh`

returns:
`Scanning installation ..

Version Info:
    Version: 0.28
    Date: 2022-07-28
    Revision: 17f8a92a6

Using qmake: qmake-qt4

Using Ruby interpreter: ruby
    Ruby library found: /usr/lib/x86_64-linux-gnu/libruby-3.0.so.3.0.4
    Ruby headers found: /usr/include/ruby-3.0.0 and /usr/include/x86_64-linux-gnu/ruby-3.0.0
    Ruby installation is in:
    - /usr/lib/x86_64-linux-gnu/libruby-3.0.so.3.0.4 (lib)
    - /usr/include/ruby-3.0.0 (headers)
    - /usr/include/x86_64-linux-gnu/ruby-3.0.0 (arch headers)
    Ruby version code is 30004

Using Python interpreter: python3
    Python library found: /usr/lib/x86_64-linux-gnu/libpython3.10.so
    Python headers found: /usr/include/python3.10
    Python extension suffix: .cpython-310-x86_64-linux-gnu.so
    Python installation is in:
    - /usr/lib/x86_64-linux-gnu/libpython3.10.so (lib)
    - /usr/include/python3.10 (includes)

Features:
    Qt bindings enabled
    Compilation caching is deactivated!
    Installation target: /home/bruno/Postdoc/job/klayout/bin-release
    Build directory: /home/bruno/Postdoc/job/klayout/build-release
    Building plugins: streamers tools 

Running qmake-qt4 ..
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu`

and not long after:
`g++ -c -m64 -pipe -std=c++11 -fvisibility=hidden -O2 -D_REENTRANT -Wall -W -pedantic -Woverloaded-virtual -Wsign-promo -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -Wno-reserved-user-defined-literal -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=30004 -DHAVE_QT -DHAVE_QT_NETWORK -DHAVE_QT_SQL -DHAVE_QT_SVG -DHAVE_QT_PRINTSUPPORT -DHAVE_QT_MULTIMEDIA -DHAVE_QT_DESIGNER -DHAVE_QT_XML -DKLAYOUT_VERSION=0.28 -DKLAYOUT_VERSION_REV=17f8a92a6 -DKLAYOUT_VERSION_DATE=2022-07-28 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DKLAYOUT_TINY_VERSION=0 -DMAKE_TL_LIBRARY -DQT_NO_DEBUG -DQT_SCRIPT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../../src/tl/tl -I/usr/include/qt4/QtDesigner -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4/QtSql -I/usr/include/qt4/QtScript -I/usr/include/qt4 -I. -I../../../src/tl/tl -I. -o tlAssert.o ../../../src/tl/tl/tlAssert.cc
In file included from ../../../src/tl/tl/tlException.h:30,
                 from ../../../src/tl/tl/tlAssert.cc:25:
../../../src/tl/tl/tlInternational.h:32:11: fatal error: QString: No such file or directory
   32 | # include <QString>
      |           ^~~~~~~~~
compilation terminated.`

In addition, -qt5 option is ignored with warning:
`*** WARNING: -qt5 option is ignored - Qt version is auto-detected now.`

Here are the versions on my system:
`qmake-qt4 --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu`
`qmake --version
QMake version 3.1
Using Qt version 5.15.4 in /usr/lib/x86_64-linux-gnu`
`gcc --version
gcc (Debian 11.3.0-5) 11.3.0`

klayoutmatthias, 2022-07-31:
Please check if a file named "QString" is available here: /usr/include/qt4/QtCore

If not, something is broken with your Qt4 installation.

However, I'd recommend using Qt5 anyway. You can try

```
build.sh -qmake qmake 
```

to make build.sh use the Qt5 version.

Matthias

bgarbin, 2022-08-08:
Something definitely seems wrong with my qt4 install and no file /usr/include/qt4/QtCore/QString was present. As a first step, I tried to fix this which does not seem to be an easy task..

I then, tried using Qt5, which does not work with very similar error. 

The installer returns this time:
```
Running qmake ..
QMake version 3.1
Using Qt version 5.15.4 in /usr/lib/x86_64-linux-gnu
```
and fails after a couple of qmake calls with error:
```
g++ -c -m64 -pipe -std=c++11 -fvisibility=hidden -O2 -D_REENTRANT -Wall -W -pedantic -Woverloaded-virtual -Wsign-promo -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -Wno-reserved-user-defined-literal -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=30004 -DHAVE_QT -DHAVE_QT_NETWORK -DHAVE_QT_SQL -DHAVE_QT_SVG -DHAVE_QT_PRINTSUPPORT -DHAVE_QT_MULTIMEDIA -DHAVE_QT_DESIGNER -DHAVE_QT_XML -DKLAYOUT_VERSION=0.28 -DKLAYOUT_VERSION_REV=17f8a92a6 -DKLAYOUT_VERSION_DATE=2022-07-28 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DKLAYOUT_TINY_VERSION=0 -DMAKE_TL_LIBRARY -DQT_NO_DEBUG -DQT_SCRIPT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../../src/tl/tl -I/usr/include/qt4/QtDesigner -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4/QtSql -I/usr/include/qt4/QtScript -I/usr/include/qt4 -I. -I../../../src/tl/tl -I. -o tlAssert.o ../../../src/tl/tl/tlAssert.cc
In file included from ../../../src/tl/tl/tlException.h:30,
                 from ../../../src/tl/tl/tlAssert.cc:25:
../../../src/tl/tl/tlInternational.h:32:11: fatal error: QString: No such file or directory
   32 | # include <QString>
      |           ^~~~~~~~~
```

I verified that /usr/lib/x86_64-linux-gnu/qt5/QtCore/QString exists.

Of additional potential interest there was (in the first call to qmake):
```
Project ERROR: Unknown module(s) in QT: designer multimedia multimediawidgets svg xmlpatterns
```
The packages that seem to match are all installed and up to date (libqt5designer5  libqt5multimedia5  libqt5multimediawidgets5  libqt5svg5  libqt5xmlpatterns5)

Many thanks,

Bruno

PS: you may want to modify the README about -qt5 option at build

klayoutmatthias, 2022-08-08:
Hi Bruno,

thanks for the note about the -qt5 option.

Regarding bookworm, I was able to do a build on a fresh docker image (debian:bookwork) using these packages:

```
apt-get update -y
apt-get install -y python3-dev ruby-dev make g++ git qttools5-dev qttools5-dev-tools qtmultimedia5-dev libqt5svg5-dev libqt5xmlpatterns5-dev
```

I do not see the `<QString>` problem :(

Regarding the include files, maybe uninstalling and re-installing the mentioned packages helps.

Matthias

bgarbin, 2022-08-09:
Hi Matthias,

Many thanks about your timely and helpful answer.

I have installed all the packages you mentioned and the compilation now started!

Here is the full list of the packages that got installed:
libqt5designercomponents5 libqt5svg5-dev libqt5webkit5 libqt5xmlpatterns5-dev  qdoc-qt5 qhelpgenerator-qt5 qt5-assistant qtattributionsscanner-qt5  qtmultimedia5-dev qttools5-dev qttools5-dev-tools

Bruno

closed 2022-08-09 by bgarbin