KLayout/klayout #1131: Donm

stefanottili, 2022-07-29:


stefanottili, 2022-07-29:
It would be nice if File->Open Recents would only list files that still exist when this menu get's opened.

Kazzz-S, 2022-07-30:
I agree with you.
I am solving this issue with my homemade macros [SamplePYAmacros.zip](https://github.com/KLayout/klayout/files/9224099/SamplePYAmacros.zip), as shown below.
This approach can apply to other "Recent..." including the layer properties, sessions, and bookmarks.

Just for your reference.
Kazzz-S

----

1. I have already deleted "Abacus.oas"
[image]
2. Hence, open error
[image]
3. Let's check the existence
[image]
4. Updated the list
[image]
5. Clean up
[image]
[image]

Undo and Redo are not possible.

Kazzz-S, 2022-07-30:
@klayoutmatthias,

I'd like to know how to show the added macro menus at the bottom of the side file list.
[image]

Warm regards,
Kazzz-S

klayoutmatthias, 2022-07-31:
@kazzz Adding menu items to these menus is not possible currently as these menus are created dynamically and before that, all items are cleared :(

Nevertheless I have implemented a fix for this ticket. As of now, non-existing files are grayed out rather than hidden. I found this slightly more intuitive as it conveys more information. In addition, there is no interference with the logic that limits the number of entries in each MRU list - if a file is only temporarily unavailable, it will not be deleted entirely, but will show up as enabled again.

I have also implemented a "clear list" menu item at the end of the list. There is no undo available for this function, but I hope, users know what they are doing :)

There has been much development in that area for the Qt-less Python module, so I'm afraid the feature will not make it into the 0.27 branch.

Best regards,

Matthias

Kazzz-S, 2022-07-31:
@klayoutmatthias,

Thank you for your prompt enhancement work. I like your idea. 
I can shorten the enlarged menu lists.
[image]

Warm regards,
Kazzz-S

stefanottili, 2022-08-01:
clear list doesn't seem to be persistent.
clear list, close klayout, run klayout again, all the cleared files are back.

klayoutmatthias, 2022-08-01:
My fault. Did not write the configuration properly. 

(did not exit either, because I did not want to spoil my recent file list :) )

Patched master as the PR is already merged.

commit 5e3358d6 (2022-08-01): Fixed #1131 fix
closed 2022-08-01 by klayoutmatthias