KLayout/klayout #1132: LEF produce pin on polygons places pin label off the shape

arlpetergadfort, 2022-07-30:
When reading in a LEF file that contains a POLYGON pin (such as an L shape), the label that is produced falls outside the shape itself.
Here is an example of this:
[image]

It appears to be a bug, I would have anticipated the pin label getting put inside the POLYGON shape.
I've attached the LEF file I used to make the image above: [POLYGON_PIN.zip](https://github.com/KLayout/klayout/files/9226990/POLYGON_PIN.zip)

Please let me know if you need anything else. Thanks.

klayoutmatthias, 2022-07-31:
Well, so far all pin polygons have been convex and taking the center of the bounding box was sufficient ...

Anyway, generating a point inside a polygon in the generic case isn't trivial. But I can make sure, the label is at a corner of the polygon.

Will this be sufficient?

Matthias

arlpetergadfort, 2022-07-31:
> Well, so far all pin polygons have been convex and taking the center of the bounding box was sufficient ...
> 
> Anyway, generating a point inside a polygon in the generic case isn't trivial. But I can make sure, the label is at a corner of the polygon.
> 
> Will this be sufficient?
> 
> Matthias

Hi Matthias,

Yes, that should be sufficient, as the pins in the PDK I derived this from puts them on the left edge and bottom edge of the shape, so I think any point on the perimeter will work.

Thanks for getting back to me on this.

klayoutmatthias, 2022-07-31:
Ok, so I'll put them here unless the pin is actually a rectangle:

[image]

Matthias

arlpetergadfort, 2022-07-31:
> Ok, so I'll put them here unless the pin is actually a rectangle:
> 
> [image]
> 
> Matthias

Matthias, that would be perfect.

Thanks.

closed 2022-08-01 by klayoutmatthias