KLayout/klayout #1137: (feature) Add support for opening .odb files from OpenROAD

luccareinehr, 2022-08-09:
OpenDB from OpenROAD implemented a new binary format for saving design databases (instead of LEF/DEF), .odb. It would be nice to be able to open these files in klayout. 

To debug designs in the OpenROAD flow, we currently need to first convert them from ODB to DEF, and only then open the design in klayout.

stefanottili, 2022-08-09:
What’s your use case ?

You are aware that OpenRoad has a really nice gui, right ?
It has access to netlist/clock tree/timing and power information that klayout has no concept of. (No scan yet afaik)

For collecting p&r data with leafcell gds/oasis and eventually tiling, I agree, it would be useful to avoid def and read odb directly.

I don’t know how stable odb is by now.
Is the license permissive enough to be able to take openroad code to add to gpl’ed klayout ?

luccareinehr, 2022-08-10:
@stefanottili I guess you're right... thought to just leave it as an idea for the future

closed 2022-08-10 by luccareinehr