KLayout/klayout #1142: Crashing on MacOS v12.5

demisjohn, 2022-08-17:
I *think* this issue started only after Upgrading from MacOS 12.4.x to 12.5.

Double-clicking KLayout.app, and it immediately pops a "KLayout crashed" notification.  Error message attached.

Tried the following with no fix:
- deleted ~/Library/de.klayout.plist,
- deleted ~/.klayout/everythign inside here
- re-downloaded Klayout 0.27.10 (the latest with a native MacOS installer).  KLayout 0.27.11 doesn't have a non-MacPort/HomeBrew/Anaconda version yet?
None of the above worked, never got Klayout to open.

@Kazzz-S any suggestions?

[2022-08-16 KLayout crash report.txt](https://github.com/KLayout/klayout/files/9355468/2022-08-16.KLayout.crash.report.txt)

Kazzz-S, 2022-08-17:
Hi @demisjohn,

Thanks for your feedback.
The primary reason is **missing Python 2.7 binaries** in macOS Monterey (12.5 is the latest).
```
:
Reason: tried: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file), '' (no such file)
: 
```
Unfortunately, I cannot build the **ST\*.dmg** package anymore due to the below reasons. [source](https://github.com/KLayout/klayout/issues/1140)
[image]

In your macOS 12.5 environment, the easiest way would be to install the **Srl.No. 35: HW\*.dmg** package, which does not need any development environment installed.

To install and run an **LW\*.dmg**, you need to install the matching development environment, that is, **Anaconda3, Homebrew, or MacPorts.**
If you wish to try this, please refer to the text files in the **xxxUser-ReadMeFirst** folder of the installer.

Kazzz-S

demisjohn, 2022-08-22:
Thanks @Kazzz-S , I'll try this - but I don't see a "serial number 35" listed on the downloads page (in MacOS section):
https://www.klayout.de/build.html

Where do I find the serial number?

demisjohn, 2022-08-22:
Oh I now see this "Monterey (experimental" version - I'll try that.
[HW-klayout-0.27.11-macOS-Monterey-1-qt5Brew-RsysPhb38.dmg](https://www.klayout.org/downloads/MacOS/HW-klayout-0.27.11-macOS-Monterey-1-qt5Brew-RsysPhb38.dmg) (Monterey (12.0)

demisjohn, 2022-08-22:
It works, thanks!

closed 2022-08-22 by demisjohn