KLayout/klayout #1152: Conda Package

curtisma, 2022-09-09:
I'd like to create a [Conda ](https://docs.conda.io/en/latest/) package for klayout to make it easy to install in a Conda environment.  I propose to add klayout as a package on [conda-forge](https://conda-forge.org/).  This will allow it to be used in a conda environment as part of [viper-forge](https://www.cascode-labs.org/viper-forge/), our effort to make a very easy to setup open-source IC design environment using Conda.

[Conda-forge](https://conda-forge.org/) is an organization that provides build, hosting, and build automation infrastructure for Conda packages.  Unless there are objections, I'd like to submit a conda packaging recipe to conda-forge.

@klayoutmatthias, would you like to be added as another maintainer of the conda package when I submit the recipe?  I don't think you would need to do anything, but it might be good so you have access to it.

mithro, 2022-09-09:
FYI - @proppy

saicharan0112, 2022-09-09:
@curtisma Is this package available in python3.7 version?

curtisma, 2022-09-09:
@proppy: Would you be interested in being added as a maintainer.

saicharan0112, 2022-09-09:
@curtisma Is there any way I can install this package locally? I just want to try for OpenFASOC
https://github.com/idea-fasoc/OpenFASOC

curtisma, 2022-09-09:
> @curtisma Is there any way I can install this package locally as you upload to the conda-forge in the meanwhile? I just want to try for OpenFASOC

@saicharan0112 Cool, I haven't created the recipe yet, but I will point you to it once it's ready, even before it's merged into a conda-forge feedstock.  I'm hoping to have a chance to put it together sometime this weekend.

saicharan0112, 2022-09-09:
> > @curtisma Is there any way I can install this package locally as you upload to the conda-forge in the meanwhile? I just want to try for OpenFASOC
> 
> @saicharan0112 Cool, I haven't created the recipe yet, but I will point you to it once it's ready, even before it's merged into a conda-forge feedstock.  I'm hoping to have a chance to put it together sometime this weekend.
> 
> I might even be able to upload the package file here.  Are you looking to run it on Linux or Windows?

Linux but I am planning to add this to the jupyter notebook first. This Klayout is the missing piece to visualize the gds file generated from one of the openfasoc generators.

klayoutmatthias, 2022-09-10:
@curtisma sure - you can add me as maintainer of course.

Honestly I have no experience with Conda recipes, but as Conda appears to be a promising deployment path on Windows I'm interesting in learning here.

What exactly is the plan? Just deploying Python packages or the full application?

Thanks and best regards,

Matthias

curtisma, 2022-09-11:
> @curtisma sure - you can add me as maintainer of course.

Great, sounds good.

> Honestly I have no experience with Conda recipes, but as Conda appears to be a promising deployment path on Windows I'm interesting in learning here.

Yeah, I'll be sure to have a windows package built

> What exactly is the plan? Just deploying Python packages or the full application?

Well, I'm thinking we should first package the full application as a package named "klayout" and then we can create a python-only package, "klayout-python".

curtisma, 2022-09-12:
So I've started the initial recipe at:
https://github.com/cascode-labs/staged-recipes/tree/klayout

and started the following PR for conda-forge:
https://github.com/conda-forge/staged-recipes/pull/20396

proppy, 2022-09-12:
@curtisma feel free to also add me as a co-maintainer

curtisma, 2022-09-12:
> @curtisma feel free to also add me as a co-maintainer

Yeah, definitely

lukasc-ubc, 2022-10-31:
Thank you folks, this is a great initiative and will help me teach courses where my students use Linux, Mac and Windows.  Would be great to have everything easy to install including the ability for KLayout to use a full Python environment.

mithro, 2022-10-31:
@lukasc-ubc - Checkout https://github.com/hdl/conda-eda and the other packaging efforts under https://github.com/hdl