KLayout/klayout #1153: Missing d25 attribute in Python API?

thomasdorch, 2022-09-10:
Hi,

I'm using the Python API for KLayout from PyPI (0.27.11) to read/write .lyt files with the db.Technology.to_xml() and db.Technology.technology_from_xml() methods and would like to set the 2.5D view options for the technology as well. There doesn't appear to be an attribute in the Technology object that corresponds to the `<d25>`, however. Is there a way to add this via the Python interface that I'm missing or would I need to manipulate the XML directly to modify this block?

klayoutmatthias, 2022-09-10:
The D25 feature was experimental in 0.27 and is going to change entirely in 0.28.

In 0.28 D25 views are generated by scripts similar to DRC scripts. These are not parts of the tech files but reside in special directories. In 0.28 KLayout recognizes a folder called "d25" and there is no corresponding tech component any longer.

A preliminary documentation of the 0.28 feature can be found here: https://www.klayout.org/downloads/master/doc-qt5/about/25d_view.html

Matthias

thomasdorch, 2022-09-18:
Thanks for the clarification. Looking forward to the 0.28 release!

closed 2022-09-18 by thomasdorch