KLayout/klayout #1154: Problems with pin labels when opening a DEF

elena-chernykh, 2022-09-12:
Hello,

I'm trying to open the following DEF file with Klayout. 

```
VERSION 5.8 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
DESIGN pins ;
UNITS DISTANCE MICRONS 2000 ;
TRACKS X 500 DO 399 STEP 1000 LAYER M6 ;
TRACKS Y 500 DO 425 STEP 1000 LAYER M6 ;
PINS 144 ;
    - GND + NET GND + SPECIAL + DIRECTION INOUT + USE GROUND
      + PORT
        + LAYER M6 ( -1500 -5000 ) ( 1500 5000 )
        + LAYER M6 ( -15000 -5000 ) ( -12000 5000 )
        + LAYER M6 ( -28500 -5000 ) ( -25000 5000 )
        + FIXED ( 30000 20000 ) N ;
    - VDD + NET VDD + SPECIAL + DIRECTION INOUT + USE POWER
      + PORT
        + LAYER M6 ( -1500 -5000 ) ( 1500 5000 )
        + LAYER M6 ( -15000 -5000 ) ( -12000 5000 )
        + LAYER M6 ( -28500 -5000 ) ( -25000 5000 )
        + FIXED ( 23250 20000 ) N ;
END PINS
END DESIGN
``` 

VDD and GND pins correspond to several polygons, but Klayout makes text labels for only one of them. I need to label each polygon. I tried to separate the first VDD pin polygon in the DEF file and Klayout labeled two polygons, the first and the last VDD:

```
    - VDD + NET VDD + SPECIAL + DIRECTION INOUT + USE POWER
      + PORT
        + LAYER M6 ( -1500 -5000 ) ( 1500 5000 )
        + FIXED ( 23250 20000 ) N ;
    - VDD + NET VDD + SPECIAL + DIRECTION INOUT + USE POWER
      + PORT
        + LAYER M6 ( -15000 -5000 ) ( -12000 5000 )
        + LAYER M6 ( -28500 -5000 ) ( -25000 5000 )
        + FIXED ( 23250 20000 ) N ;
```

I find it strange. I think this is an incorrect solution to the problem. Is there any way to fix the pin marking problem?

klayoutmatthias, 2022-09-20:
You see that is on my TODO list:

[image]

But I recall there was some discussion about the proper way. I'm used to one label on a net, regardless of how many shapes are there. LVS netlisting is happy with a single label. Placing a text on every shape seems like a lot of redundancy to me.

The DEF reader allows attaching net names as properties to every shape. Maybe that helps?

I'd leave the decision to the user community. So the discussion is open.

elena-chernykh, 2022-09-21:
This problem is because I use a DEF-file to get a GDS-file using the Layout.read(string filename) function, and after that I generate a LEF file from GDS using the follow python script. Therefore, I need each polygon with pin to have its own label. 
Maybe in this case there is some alternative solution?
[gds2lef.txt](https://github.com/KLayout/klayout/files/9618710/gds2lef.txt)

elena-chernykh, 2022-09-26:
Thank you for the proposed solution to my problem! It can really help me. 

However, it would be great if I could get the label of each polygon of pins from GDS obtained using the following code: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/klayout/def2gds.py. 
Is it possible?

klayoutmatthias, 2022-09-26:
I've shown above how you can use another property to tag pins (bottom example, pin property key is 2). That should allow you to retrieve all polygons by pin name.

I guess by doing so you can achieve the same results than the Python code although I did not analyze it thoroughly.

Matthias

elena-chernykh, 2022-10-01:
Actually the point of previous message was that we're trying to implement LEF abstract generation for Openlane flow with KLayout. GDS files in this flow are generated by def2gds.py mentioned above. I could try to make pull-request to Openlane with addition of properties to this script, but I'm not sure it will be accepted as it makes GDS files more different from ones produced by Magic (primary Openlane tool for this operation as for now). So in my opinion it will be more logical to mark all appropriate shapes with pin name.

klayoutmatthias, 2022-10-02:
I wasn't suggesting to change def2gds.py. Instead you can directly read DEF and produce LEF from that in a single step.

klayoutmatthias, 2022-12-06:
Any comments? I'd close this ticket otherwise.

Matthias

closed 2023-08-09 by klayoutmatthias