KLayout/klayout #1159: Incorrect usage of std::unique_ptr in tl::to_local()

EugeneZelenko, 2022-09-21:
I encountered Valgrind's complains about mismatched `delete / delete []` when just importing `pymod` from 0.26.4 (but `std::auto_ptr` was used back then.

`std::unique_ptr<char> buffer (new char [MB_CUR_MAX]);` should be `std::unique_ptr<char[]> buffer (new char [MB_CUR_MAX]);`, per https://en.cppreference.com/w/cpp/memory/unique_ptr.

klayoutmatthias, 2022-09-22:
Thanks for the remark. It's fixed in the "wip" branch.

But the address sanitizer remark is interesting. Where can I find this library?

Matthias

EugeneZelenko, 2022-09-22:
Address Sanitizer (as well as other sanitizers) comes with GCC and LLVM/Clang. Complete instrumentation requires `-fsanitize=address` in compiler command line.

To detect issues that originates in incorrect standard API usage you could use:

```
export LD_LIBRARY_PATH=<your GCC/LLVM path>/lib:${LD_LIBRARY_PATH}
export LD_PRELOAD=libasan.so
```

and than run KLayout (or any other application).

By the word, this was only issue detected by Address Sanitizer in such way.

closed 2022-10-26 by klayoutmatthias