KLayout/klayout #1162: Loading layer properties without a layout view?

donn, 2022-09-27:
* Version: KLayout 0.27.11

In our Klayout stream-out script for OpenLane, we load the layout as follows:

```
    tech = pya.Technology()
    tech.load(tech_file)
    layout_options = tech.load_layout_options
    layout_options.lefdef_config.macro_resolution_mode = 1
    layout_options.lefdef_config.lef_files = [lef_file]

    # Load def file
    main_layout = pya.Layout()
    # main_layout.load_layer_props(layer_props_file)
    main_layout.read(in_def, layout_options)
```

However, `load_layer_props` doesn't seem to exist unless we use LayoutView, which we do not so far in this script. Is there any way around this?

klayoutmatthias, 2022-09-30:
Hi @donn,

Basically layer properties are part of the LayoutView configuration, hence there is no such thing inside MainWindow when no LayoutView exists.

But I wonder why you need layer properties on stream out. I suppose "stream out" means DEF to GDS conversion. Layer properties are not involved here - these only control display of layout. The stream readers and writers do not use layer properties.

The DEF reader can use a **layer map** file which controls the translation of DEF layer names to GDS layer/datatype numbers. Maybe you refer to this feature?

Matthias

donn, 2022-10-27:
Sorry for the late reply- yeah, that'll do. Thank you so much.

closed 2022-10-27 by donn