KLayout/klayout #1163: Illegible overlapping Cell names

demisjohn, 2022-09-29:
When multiple cells have overlapping origins, the title of the Cell is very difficulty to read.  Although text is outlines in white, and clicking on the cell cycles which is in front, this still is rather difficult to read.

This is only 2 names overlapping, more gets even more difficult:
[image]

May I suggest to grey the background texts, and only the frontmost is black - perhaps that would make it easier to read.  Hard to know unless you try it though.

This is a GUI issue that constantly crops up, and is difficult to circumvent. The best I've figure out is to click and watch the status bar which tells me what I'm clicking on - and make a mental list of the objects' names.

klayoutmatthias, 2022-09-29:
There are options to configure this already.

First you can configure the way, cell boxes are drawn (here: "Cell box colors")

[image]

which gives you:

[image]

Then you can configure the selection color to some contrast color (here: "Marker color"):

[image]

Which gives you that:

[image]

and

[image]

Matthias

closed 2022-12-06 by klayoutmatthias

demisjohn, 2023-02-02:
@klayoutmatthias can we make your above settings the default for the app? That would indeed solve this issue - with the current selection’s text always being legible.

demisjohn, 2023-02-03:
That’s a great idea - I’ll make & distribute an RC file for our users.